ARM-software/psa-arch-tests #27: Uninitialised local variable fakes the test 012

zaya-mc, 2019-01-08:
In the psa_hash_update_invalid_handle() function in the apt-tests/dev_apis/crypto/test_c012/test_c012.c file, 
local "psa_hash_operation_t  operation;" function is not initialised so dummy values in the stack may fake the test case and negative test can pass. (In our case, test passed but it must be failed). 

So, we think we need to zero-ise this variable. 

`psa_hash_operation_t    operation = { 0 };`

gowthamsiddarthd, 2019-02-04:
A function call with any uninitialized operation (not initialized by psa_hash_setup or operation = { 0 }) to  psa_hash_update function should fail. The implementation should be able to detect this and send PSA_ERROR_BAD_STATE

zaya-mc, 2019-02-04:
Hi gowthamsiddarthd,  yes you are right but in the PSA there is an option called TEST_COMBINE_ARCHIVE, which builds all test suites/cases into single executable so if the previous test case initialises a psa_hash_operation_t correctly and exit from the function (like exit from test case) it leaves valid values in the stack, and if the valid values somehow maps to uninitialised local variable in the next test case (because it fills local variable the values from the stack), it may fake the test suite. That was what we saw. (random values were perfectly working for hash operation). 
After latest changes we dont see this issue,  but still there can be a risk. (Probably you dont build it with TEST_COMBINE_ARCHIVE). 
I am happy to close this ticket, if we see same problem, we can report it again.

athoelke, 2019-02-05:
@gowthamsiddarthd  - it is not possible to guarantee that an unitialised C variable is invalid because the compiler will do nothing to the storage memory for the variable. As @zaya-mc notes, if the memory previously had values that look like a valid operation state then the PSA Crypto implementation _cannot tell that this is a new operation object_.

So although an unitialised operation object would _normally_ be detected as `PSA_ERROR_BAD_STATE`, it will not _always_ be. It is therefore **required** that users of the PSA Crypto API **always** initialise operation objects before use:
* in version 0.1 of the API this is done by using a suitable `setup()` function
* in version v1.0-beta of the API this must be done using one of the `*_INIT` macros or `*_init()` initialiser functions.

jaypit02, 2019-02-05:
@athoelke  I believe, Test is doing negative check where it expects API to return PSA_ERROR_BAD_STATE value for uninitialised operation and it was deliberate to not to call init functions before the API.

But as @zaya-mc pointed out, there is slight chance that uninitialised operation value might be valid for the API sometimes. And when it will hit as valid, the test check will fail for the wrong reason.

@gowthamsiddarthd, let's fix this by initialising the variable with value which can be hard to hit. Please cross check similar test scenario for other API too.

I hope, this helps.

jaypit02, 2019-02-19:
The fix for this issue is merged with master as part of https://github.com/ARM-software/psa-arch-tests/pull/64 

Hence, marking this close.

closed 2019-02-19 by jaypit02