ARM-software/psa-arch-tests #79: TFM and PSA Test-Suite are not synchronized

AndreyButokNXP, 2019-04-01:
Hello,
TFM and PSA test suite are using different PSA Dev. API.
Please compare: tfm\interface\include\psa_crypto.h with psa-arch-tests\api-specs\include\psa\crypto.h

Guess,TFM uses newer version of specification. So the PSA Test-suite is obsolete and may be not used for testing the TFM latest version.
Please do regular synchronization with TFM API.
Thanks,
Andrej Butok

adeaarm, 2019-04-01:
HI @AndreyButokNXP , TF-M uses a version of the API which is marked `0.1.0b`. The psa-arch-tests, on the master branch, have moved to use a newer version of the API, while on the branch marked `ew_beta0` they use the version of the API compatible with the one TF-M uses, hence TF-M PSA API compliance needs to be tests using the `ew_beta0` branch. Work is ongoing on TF-M side to move to newer versions of the API.

jaypit02, 2019-04-03:
Closing this as query is answered. Feel free to reopen if there are questions.

closed 2019-04-03 by jaypit02

AndreyButokNXP, 2019-04-03:
Cygwin, VirtualBox and etc. is not a solution for us.
We are going to add the TestSuite to our MCUx SDK, so Cygwin is not acceptable.

jaypit02, 2019-04-04:
@AndreyButokNXP 

I think, your comment is for https://github.com/ARM-software/psa-arch-tests/issues/80 issue. Let's have related conversation there.