ARM-software/psa-arch-tests #80: Windows host operation system?

AndreyButokNXP, 2019-04-01:
Hello,
Are you going to support the Windows 10 as the host operation system?
Now, it is limited to Ubuntu only.
Thanks,
Andrej

jaypit02, 2019-04-03:
@AndreyButokNXP ,

Thanks for reporting this. The support is already available. You can use Cygwin to build PSA test suite on Windows platform. 
This information is missing in the documentation right now. The documentation will be updated to codebase along with other future changes.

AndreyButokNXP, 2019-04-04:
Cygwin, VirtualBox and etc. is not a solution for us.
We are going to add the TestSuite to our MCUx SDK, so Cygwin is not acceptable.

As a solution, we are going to add missing files to projects, so the file generation (via bach, perl,gcc) is not required.

kotegowder, 2019-06-12:
@AndreyButokNXP 

We are in the process of adding support of generating Makefiles for Windows and Linux using CMake.
On Windows cmd.exe or Eclipse debugger (with make and cmake utilities installed before hand) can be used to build the project.
On Linux it works the same way as earlier.

You can expect the support to come in next release.

kotegowder, 2019-07-25:
@AndreyButokNXP 

We enabled CMake build support in master now.
This would let you build test suite on Linux and Windows hosts.

Let us know whether we can close this issue. 

- Vinay

AndreyButokNXP, 2019-07-25:
Yes, you may close it. If any issue, I will submit a new one ;)

closed 2019-07-25 by jaypit02