ARM-software/psa-arch-tests #81: TF-M ARM GCC PSA Test 014 fail

nxf49431, 2019-04-25:
Hi,

I use “7 2018-q2-update” ARMGCC compiler. The PSA Tests code works when it is compiled by ARMGLANG, but a fail occurs when the same code is compiled by ARMGCC.

Issue:

When TEST: 414 runs, I’ve got secure violation fault. See report bellow:

TEST: 414 | DESCRIPTION: Optional APIs not supported check
[Info] Executing tests from non-secure
Optional PS APIs are not supported.
[Check 1] Call to create API should fail as API not supported
[Check 2] Create valid storage with set API
[Check 3] Call to set_extended API call should fail
[Check 4] Verify data is unchanged
Oops... Secure fault!!! You're not going anywhere!

Reason:

The valtest_entry_p014 is overwritten by read_buff variable and this caused this fail.

Workaround:

When I changed read_buff declaration in test_p014.c from static uint8_t read_buff[TEST_BUFF_SIZE/4]  = {0};  to static uint32_t read_buff[TEST_BUFF_SIZE/4]  = {0};  the code is running correctly. But I think, that it should be officially fixed.

Best Regards

Stanislav

chetansinghrathore, 2019-04-29:
In the psa_get API at check#4,  request to read 16 octets is made but the read buffer was of size 4 octets only, which might have resulted in the secure fault.

Fix is provided and PR is raised.

chetansinghrathore, 2019-04-29:
@nxf49431 : Fix is provided and merged in github.
Please close the issue, if test run successfully.

jaypit02, 2019-04-29:
Closing this since fix is provided. Feel free to reopen if you face any problem.

closed 2019-04-29 by jaypit02