ARM-software/psa-arch-tests #82: test_i063 from secure side

jeffrey4913, 2019-04-28:
Hi, 
I have gone through the IPC testcase i063(secure side), I feel there are some problems here, please confirm.
Let me briefly describe the process:  

client SP                                                           server SP
    connect(SERVER_UNSPECIFED_MINOR_V_SID,)   ----------->
                                              <----------           doorbell(Notify client partition to make SERVER_SECURE_CONNECT_ONLY_SID connection request)
    connect(SERVER_RELAX_MINOR_VERSION_SID,)  ----------->         
                                              <---------            reply SERVER_UNSPECIFED_MINOR_V_SID
                                              <---------            reply SERVER_RELAX_MINOR_VERSION_SID
    get_secure_test_result(call/close)        ---------->
                                                                    wait for SERVER_SECURE_CONNECT_ONLY_SID

The client SP handle the doorbell signal after get_secure_test_result, but the server sp handle the result after it get SERVER_SECURE_CONNECT_ONLY_SID signal..

jaypit02, 2019-04-29:
Hi @jeffrey4913,

Thanks for your query.

It seems, you are thinking of dead-lock situation where test wouldn't be able to get secure test result. 

I think, it won't go into dead-lock as there are there are parties involved in the scenario.  I capture the whole scenario in below table with expected test sequence.

1. NS client which executes test_i063.c code and wait for secure test result
2. S Server partition (test_supp_i063.c)
3. S Client partition  (client_partition.c)

NS Client | Server SP | S Client SP
---------  | ---------  |  ----------
 connect(SERVER_UNSPECIFED_MINOR_V_SID,) | _ | _
 _ | doorbell(Notify client SP to make SERVER_SECURE_CONNECT_ONLY_SID connection request) | 
_ | _ |  connect(SERVER_SECURE_CONNECT_ONLY_SID)
_ | reply SERVER_UNSPECIFED_MINOR_V_SID|
connect(SERVER_RELAX_MINOR_VERSION_SID) | _ | _
_ | reply SERVER_RELAX_MINOR_VERSION_SID | _
get_secure_test_result(call/close) | wait( SERVER_SECURE_CONNECT_ONLY_SIG) | _
_ | reply (SERVER_SECURE_CONNECT_ONLY_SID) | _
_ | Send secure result to NS client | _

> The client SP handle the doorbell signal after get_secure_test_result, but the server sp handle the result after it get SERVER_SECURE_CONNECT_ONLY_SID signal..

Yes, this seems possible sequence and high depend on partition scheduling policy.  However i'm not expecting test to fail even if this hits. In worst case, client partition won't be able to mimic irritator behaviour which test scenario is expecting.

Let me know if this helps you or I have misunderstood something.

jaypit02, 2019-05-06:
Please feel free to reopen if you need further information

closed 2019-05-06 by jaypit02

jeffrey4913, 2019-05-06:
Hi jaypit02,
Thanks for you quick reply. 
The IPC APIs are blocked until the the function return(The behavior is defined by PSA-FF). So the s client SP will be blocked in the get result process, which has nothing to do with the scheduling. The s client SP would have no change to handle the doorbell signal in current implementation.

jeffrey4913, 2019-05-06:
@jaypit02 , maybe i have no right to reopen the case.

jaypit02, 2019-05-06:
Hi @jeffrey4913,

It is NS client who executes get result function and will be blocked in get result process. The S client is free and wait for signals in main while loop. So I think, S client SP should be able to handle doorbell signal.

jaypit02, 2019-05-06:
test_i063 executes client tests (available in test_i063.c) only from non-secure. You can refer it in ff/ipc/test_i063/test_entry.c file.  And S client SP(in client_partition.c) waits through  psa_wait API and will be get unblocked when it receives the PSA_DOORBELL signal. test_i063 doesn't expect to the S client SP to execute client test dispatcher code for it. 

Did you come to see the problem by executing the test?

athoelke, 2019-05-09:
At the interesting point in the scenario:
* NSPE is _blocked_ waiting for a response to the `get_secure_test_result()` request sent to Server SP
* Server SP is _blocked_ waiting for `psa_wait()` to return with the `SERVER_SECURE_CONNECT_ONLY_SIG` signal, which depends on the Client SP to make a connection request.
* Client SP is _ready to run_ because the doorbell signal that it was waiting for has been asserted by the Secure SP

Even though NSPE and Secure SP are _blocked_, the scheduling rules in the PSA FF (section 3.2.4) require that the Client SP is executed as it is _ready to run_. So there is no deadlock as the Client SP will make the connection request, which will make the Secure SP _ready to run_.

jeffrey4913, 2019-05-20:
Hi athoelke, jaypit02,
Thanks for your reply. There is no dead lock in origial.
Maybe I didn't make myself clear. 

I made some minor modifications to the test case for I063.
 I add the following code add test_entry.c hope to run it from secure side. The the dead lock happen.
I will remove it from my test.
//////////////////////////////////////////////
status = val->switch_to_secure_client(TEST_NUM);

jaypit02, 2019-05-20:
@jeffrey4913, Thanks for confirming your understanding. val->switch_to_secure_client(TEST_NUM); API call shouldn't be there as client Partition has different role to play for this test.