ARM-software/psa-arch-tests #86: Test case panic in server side of i003/i018/i019

raymao7703, 2019-05-21:
Hi,

In the service code of test cases i003/i018/i019, it tries to access the memory of a NULL pointer and causes panic.
----------------------
For example, in function "server_test_psa_set_rhandle" of test_supp_i003.c:
int           *num = NULL;
......
*num = 5;
----------------------

Moreover, in function "server_test_psa_set_rhandle_with_null_handle" of test_supp_i019.c, line54:
psa_set_rhandle(PSA_NULL_HANDLE, num);
I think it should be something as below:
psa_set_rhandle(PSA_NULL_HANDLE, &num);

Regards,
Raymond

jaypit02, 2019-05-21:
Hi @raymao7703,

Thanks for reporting this. I comprehend, this is test issue. We have plan to push the fix in our next psa-arch-test release. Let us know if you need patch now.

jaypit02, 2019-06-06:
The fix for this issue is provided as part of PR:https://github.com/ARM-software/psa-arch-tests/pull/94

closed 2019-06-06 by jaypit02