ARM-software/psa-arch-tests #87: psa_call with non-NULL outvec but a 0-size in i051/i054

raymao7703, 2019-05-21:
Hi,

In test case i051/i054, it is supposed to check the outvec.base by psa_call.
However, the outvec size should be 1 instead of 0.
----------------

Line 118 of test_i051.c and line 75 of test_i054.c 
Change from:
psa->call(handle, NULL, 0, outvec, 0);
to:
psa->call(handle, NULL, 0, outvec, 1);

Regards,
Raymond

jaypit02, 2019-05-21:
Hi @raymao7703,

Thanks for reporting this. Yes this is a bug and we had found it and fixed it internally.  We are planning to push the fix for these tests as part of our next release in few weeks time. Let us know if you need patch now.

jaypit02, 2019-06-06:
The fix for this issue is provided as part of PR:#94

closed 2019-06-06 by jaypit02