ARM-software/psa-arch-tests #88: psa_write with incorrect size parameter in server side of i057

raymao7703, 2019-05-21:
Hi,

In the test case of i057, from client side, it is passing outvec:
   psa_outvec outvec[1] = {{&data, sizeof(data)}};
   status_of_call =  psa->call(handle, NULL, 0, outvec, 1);

However, in server side, it is trying to write buffer with invec size:
   psa_write(msg.handle, 0, (void *)buffer, msg.in_size[0]);

I think this is a mistake, and it should be something like below:
   psa_write(msg.handle, 0, (void *)buffer, msg.out_size[0]);

Regards,
Raymond

jaypit02, 2019-05-21:
Hi @raymao7703,

Thanks for reporting this. I comprehend, this is test issue. We have plan to push the fix in our next psa-arch-test release. Let us know if you need patch now.

jaypit02, 2019-06-06:
The fix for this issue is provided as part of PR:#94

closed 2019-06-06 by jaypit02