ARM-software/psa-arch-tests #91: Test 411 and 409 are failed with the latest TFM

AndreyButokNXP, 2019-05-23:
Hello,
Since the latest changes added to TFM, two tests of  PSA Protected Storage Suite are failed.
`
******************************************
TEST: 409 | DESCRIPTION: Invalid Arguments check
[Info] Executing tests from non-secure
[Check 1] Call set API with NULL pointer and data length 0
        Failed at Checkpoint: 1
        Actual: 0
        Expected: 8
TEST RESULT: FAILED (Error Code=0x1)

******************************************
TEST: 414 | DESCRIPTION: Optional APIs not supported check
[Info] Executing tests from non-secure
Optional PS APIs are not supported.
[Check 1] Call to create API should fail as API not supported
        Failed at Checkpoint: 2
        Actual: 0
        Expected: 5
TEST RESULT: FAILED (Error Code=0x1)
`
I believe, it is related with the latest improvements added to TFM, so the tests are obsolete.

Thanks,
Andrej Butok

chetansinghrathore, 2019-05-24:
Test-9 failed with latest TF-M storage implementation which is updated to beta-2 spec version. Test is also updated and now passing. Test14 was failed as UID value created in test9 didn't get deleted due to its failure.

Now all test are passing
PR#92 created

AndreyButokNXP, 2019-05-27:
Hi, 
so far, there is no fix on the ew_beta0 branch which used for TFM certification/testing.

AndreyButokNXP, 2019-05-28:
BTW: What branch must be used for the TFM PSA API certification now?
So far, the ew_beta0 branch was used.

jaypit02, 2019-06-06:
Hi @AndreyButokNXP,

If you are using latest TF-M code which is TF-Mv1.0-RC1 tag, we recommend to use our latest v0.9 PSA APIs test suite release for the PSA API certification. As we don't know the level of TF-M code changes that you are using, we would advice to consult with TF-M team on this regard.

And the requested test fixes have been merged as part of #93.

closed 2019-06-06 by jaypit02