ARM-software/psa-arch-tests #96: text_entry.c file compilation issue in Keil/uVision

AndreyButokNXP, 2019-06-25:
Hello,

Every PSA test has own test_entry.c. It means a huge number of files with the same name in one project. It causes a compilation problem in the latest version of Keil IDE v5.27.1. 
Probably, it is trying to compile files in parallel, and this is causing the test_entry<number>.o clashing at some moment.

The workaround, which works, is renaming of all test_entry.c files to test_entry_<test_id>.c.

Could you do the same for the original source code?

Thanks,
Andrej Butok

jaypit02, 2019-06-26:
Hi @AndreyButokNXP,

Thanks for reporting this.

I'm assuming that you are not using PSA test suite build flow because it compiles the each tests sequentially. Will it be possible for you to help us on reproducing the issue at our end before we provide fix?

AndreyButokNXP, 2019-06-27:
Hi,

We are compiling and run all tests in one project.
IDE: uVision/Keil v5.27.1 - should be the latest one (using default compiler version 6).
Platform: LPC55S69 (M33)
The project has 60 files with the same name "test_entry.c".
It causes an error during compilation - "duplicated <>.o" file, so it is automatically removed.

Solutions which worked:
1) reducing number of tests <50
2) renaming test_entry.c to test_entry_<test_id>.c
We have choose the (2) option.

 BTW: It looks like an issue of IDE, but having the same name for files with a different content is bad practice anyway.

Hope this is enough,
Thanks,
Andrej Butok

jaypit02, 2019-07-04:
Hi @AndreyButokNXP,

Thanks for providing the information. We will use this information to reproducing the issue and providing the solution for it in couple of week time. Let us know if this is needed before it.

Thanks,
Jay

jaypit02, 2019-07-11:
Hi @AndreyButokNXP,

PR #98 is created to add test-id to test_entry.c file name. Let me know if this solution works for your project.

Thanks,
Jay

AndreyButokNXP, 2019-07-11:
Hi Jay,

Yes, I have added absolutely the same fix.

Thanks,
Andrej

jaypit02, 2019-07-15:
The fix is available in the master. Hence closing this.

closed 2019-07-15 by jaypit02