ARM-software/psa-arch-tests #105: TFM and PSA Tests suite API conflict

AndreyButokNXP, 2019-08-05:
Hello,
In the latest commits were done change to PSA API, for example, psa_ps_info_t => psa_storage_info_t, psa_ps_uid_t => psa_storage_uid_t etc.
But it is causing compilation errors with the latest TFM, as it does not have these type of definitions.
Is it known issue? Is this issue of TFM?

Used branches:
- psa-arch-tests: the latest "master"
- trusted-firmware-m: the latest "master"

gowthamsiddarthd, 2019-08-05:
Hi @AndreyButokNXP,

In the latest commit we have updated the storage test suite to support the PSA Storage v1.0.0 release specification. Currently TF-M supports 1.0-Beta3 version of the spec and doesn't have the support for 1.0.0 release implementation of PSA storage spec.

This is a known issue and expected to have compilation errors because of the spec mismatch.
I would suggest you to use the release branch of psa-arch-test which should be compatible with the TF-M master.

Regards,
Gowtham

AndreyButokNXP, 2019-08-05:
Thank you for the confirmation.
So, switched to the next to last commit (c80681ed7c7f3e2cbf02ded1ef2464ba2ca7ccd5), which is OK.

prasanth-pulla, 2019-08-12:
Closing this issue as it is expected that release branch is used for released and aligned specifications.

closed 2019-08-12 by prasanth-pulla