ARM-software/psa-arch-tests #119: 64-bit HOST_GCC

adrianlshaw, 2019-09-20:
If you use HOST_GCC toolchain and the host happens to be 64-bit, then there are issues with the m32 flag passed in the CFLAGS and linker arguments. Ideally if HOST_GCC is used then m32 shouldn't be passed.

jaypit02, 2019-09-25:
Hi Adrian,

Thanks for this, It's helpful.

As you can see in https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/sw_requirements.md, the supported toolchains are:
- GNU Arm Embedded Toolchain 6.3.1
- Arm Compiler v6.7
- GCC 5.4.0 32-Bit (Linux Host)
 - MinGW 6.3.0 32-Bit (Windows Host)

We are yet to investigate the PSA test suite support requirement for 64 bit host machine and use of 64 bit HOST_GCC in that case. However, we have recorded your feedback internally and we will relook at it when we plan to go for this support.

jainvikas8, 2019-10-23:
Hi, @jaypit02 since we have https://github.com/ARM-software/psa-arch-tests/pull/113 support now... I think it would be ideal to add 64-bit support.

jaypit02, 2020-02-27:
In the investigation, it came out that PSA API test suite is targeted for M profile and doesn't see the need of supporting 64 bit platform in near future. We will plan to take this up when we come across for need. Hence Closing this.

closed 2020-02-27 by jaypit02