ARM-software/psa-arch-tests #120: Attestation test suite fails if token size bigger than 0x200

tamasban, 2019-09-27:
- Due to the support of multi image boot (TF-M commit_id: 7384ee7) the size of the attestation token has increased, because there is one item for S and NS image in the SW_COMPONENTS array separately. Size exceeds the 0x200 value.

- In api-tests/dev_apis/initial_attestation/test_a001/test_a001.c a fixed size (0x200) buffer is allocated for holding the attestation token.

- In line 47 the expected size of the token is retrieved from attestation service (bigger then 0x200) and in line 62 this size is passed to the service in get_token()  call. As a result a buffer overflow happens, because the actual buffer size is smaller then the reported size. 

- I tried to fix it to increase the TOKEN_SIZE macro in test_a001.h to be able to accommodate the increased size token, but did not help. During token parsing a Hard_Fault was triggered at a random point in the code, when some variables was pushed to the stack. Might stack limit exceeds? I did not verify this theory.

- As a workaround I disabled the multi image support and reverted back to single image mode (combined S_NS) in this case test suite executed as expected. Adding this to cmake command line: -DMCUBOOT_IMAGE_NUMBER=1

gowthamsiddarthd, 2019-10-09:
The fix for the issue is submitted here: https://github.com/ARM-software/psa-arch-tests/pull/127

gowthamsiddarthd, 2019-10-11:
Closing this issue as the fix has been check-in.  Please feel free to reopen with further info.

closed 2019-10-11 by gowthamsiddarthd