ARM-software/psa-arch-tests #121: Omission of key-id from attest token is not supported

tamasban, 2019-09-27:
The specification of initial attestation token does not require to have an explicit key-id in the unprotected part of the attestation token. Furthermore according to the COSE standard the unprotected header can be just an empty map (just single 0xA0 byte).

Take a look on this change in TF-M:
https://review.trustedfirmware.org/#/c/trusted-firmware-m/+/1899/

In TF-M there is compile time switch to remove it from the token. By default the Release build omits the key-id and the debug build include it. Both version in complaint with the COSE standard and IAT specification.

gowthamsiddarthd, 2019-10-11:
The fix for the issue is submitted here: #127

gowthamsiddarthd, 2019-10-11:
Closing this issue as the fix has been check-in.  Please feel free to reopen with further info.