ARM-software/psa-arch-tests #125: PSA Storage test 406 Flags not supported test not cognisant of flags behaviour

simonqhughes, 2019-10-02:
The test is incorrect in that it sets a flag bit when an object
is created, and expects the bit always to be set in the flags retrieved by
psa_xx_get_info(). The specification defines this is not always the case.

The following outlines psuedo-code for this test:

```
 set create_flags = 0x800000000
  while (create_flags != 0)
    {
      psa_ps_set()
      psa_ps_get()
      // check xxx
      psa_ps_get_info()
      // check flag retrieved is the same at the value used to set
      psa_ps_remove()
      create_flags >>= 1
    }
```
```
However:
- This test doesn't respect the behaviour of defined flags:
       - PSA_STORAGE_FLAG_WRITE_ONCE
           - If this bit is set then the object cannot be deleted.
             psa_xx_remove() will fail but the test current checks the
            object can successfully be removed.
       - PSA_STORAGE_FLAG_NO_CONFIDENTIALITY
           - The psa_trusted_storage_linux implementation always stores
             files with authentication (because files are stored with
             ecryptfs). Therefore if _NO_CONFIDENTIALITY is requested then
             its accepted but files are still stored confidentially. This
             is allowed by the spec. The spec also says that if this
             happens the the _NO_CONFIDENTIALITY flag should not be
             reported with get_info().
       - PSA_STORAGE_FLAG_NO_REPLAY_PROTECTION
           - This flags should be observed with the psa_xx_get_info()
             after creation.
  - Only 3 of the flag bits are defined. The specification should be
    modified to say something 1) the undefined flag bits
    are reserved and the operation of the flags is undefined, or 2)
    if flag bit is set then the value should be retained with the object
    i.e. the behaviour being tested for here.

 The solution if to not test for the following flags:
   - PSA_STORAGE_FLAG_WRITE_ONCE
   - PSA_STORAGE_FLAG_NO_CONFIDENTIALITY
```

The following link shows a commit if a forked repo containing a work around:

```
 Fix PSA Storage test 406 - flags not supported test.

The following provides more information on this commit:
- The above test implements the setting and getting of
  object creation flags without regard for the behaviour
  of the flags. This fix skips set()/get_info() tests
  for the affested flags.
- The test should be fixed to better reflect the
  behaviour of the flags.
```

https://github.com/ARM-software/psa-arch-tests/commit/acad60c96976aff5a201db2aec6b2c1033234506

kotegowder, 2019-10-03:
Thank you for reporting the issue.

The latest code (in align with 1.0.0) from test_s006.c do not create the storage against PSA_STORAGE_FLAG_WRITE_ONCE that means we are not attempting for psa_its_remove. 

PSA_STORAGE_FLAG_NO_CONFIDENTIALITY has to be handled against the below line from spec:

"When reporting meta data, psa_ps_get_info() should report the actual protection level applied and not the requested level." (which is applicable to psa_its_get_info() as well)

So as you explained with your implementation psa_its_get_info() will not have bit set when psa_its_set() is requested with PSA_STORAGE_FLAG_NO_CONFIDENTIALITY flag. 

Since the implementation creating storage with higher security level - this has to be accepted. 

We will make changes for this.

kotegowder, 2019-12-09:
This is addressed as part of github patchset.

closed 2019-12-09 by kotegowder