ARM-software/psa-arch-tests #126: `caller_security_t` value never referenced checked.

jainvikas8, 2019-10-02:
In `api-tests\val\common\val.h` we have **caller_security_t** being defined

`typedef enum {
    CALLER_NONSECURE = 0x0,
    CALLER_SECURE    = 0x1,
} caller_security_t;`

But I don't see these being referenced checked anywhere in the tests. For example if we look at the definition of `int32_t psa_hash_compare_test(caller_security_t caller);`, we don't see **caller** variable being used?

What is the purpose of this? If we don't see any foreseeable purpose of this can we remove it?

jaypit02, 2019-10-02:
Hi Vikas,

caller_security_t is being used by the PSA FF(IPC) tests where test_i[x].c code is expected to be repeated for secure and non-secure side. 

We can't remove this variable as test framework is generic for FF and developer APIs tests. Moreover, this variable can also help to scale developer API tests to run at secure side easily in future. However it is not the requirement right now

jaypit02, 2019-10-08:
I think, the query has been answered. Hence closing this. Please feel free to reopen with further info.

closed 2019-10-08 by jaypit02