ARM-software/psa-arch-tests #131: NVMEM area in Musca_B1 ?

AlamyLiu, 2019-10-23:
Is there a document or something to describe/clarify the NVMEM area used by Musca_B1 ?

If my study is correct...
---- PSA ----
nvmem.0.start = 0x20060000;
nvmem.0.end = 0x200603FF;
---- TF-M ----
0x0a060000 / 0x0007f800  NS_CODE
0x1a000000 / 0x0003f480   S_CODE
0x20040000 /    0x40000  NS_DATA (256 KB)
0x30000000 /    0x40000   S_DATA (256 KB)
  0x30000000  BOOT_TFM_SHARED (Shared data between bootloader & runtime)
  0x30000000 /  0x80000 BL2_DATA

The NVMEM is right in the middle of NS_DATA, and I found nothing related to it in
  musca_ns.ld (gcc)
  musca_ns.sct (armclang)
to protect that area.

May I know how the nvmem start/end is calculated ?

gowthamsiddarthd, 2019-11-13:
Hi @AlamyLiu,

In the test suite, NVMEM are is used to store and preserve the test information across reboots. This requirement is mentioned https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/porting_guide_dev_apis.md

The NVMEM can be a flash memory, external memory or in this case a memory in RAM which is not used and will not be initialized during the boot sequence. The details of the NVMEM area mentioned in the target.cfg which is target specific. We expect the target users to fill the required data in the target.cfg. 

In this case, the data is provided by the TF-M. Hence we expect the TF-M to protect the NVMEM region. PSA test suite requires 1KB of NVMEM to preserve the test status across reboots. We do expect that this memory does not overlap with the one that is used by the firmware. We might need to recheck the address, but it is purely relies on the specific platform owner.

Regards,
Gowtham

AlamyLiu, 2019-11-13:
Hi ARM,

Thanks for the explanation.
I'm more curious about how the TF-M team came up with that address, to learn if there was other concern to be aware of.
Since the current setting works, and as long as it's protected/reserved for PSA. I'm fine with that.

Regards,
Alamy

closed 2019-11-13 by AlamyLiu