ARM-software/psa-arch-tests #150: It's not possible to enable the [1-10] ITS and PS tests in one project.

AndreyButokNXP, 2020-03-02:
ITS and PS share source code for the first 10 tests.
If defined ITS_TEST, test_s001-test_s010 are used for ITS, otherwise if defined PS_TEST they are used for PS. But it's not possible to run 1-10 tests simultaneously for PS and ITS.
We are compiling all PSA source codes in one project, NOT as a library per test suite.
Could you fix it? Probably, the best solution do not mix user code for PS and ITS.

kotegowder, 2020-03-02:
Hi Andrej,

The current design of psa-arch-tests for ITS and PS (for the fact of any dev APIs), one can run any one (-DSUITE )at a time.

Since the logic remains the same for PS and ITS (tests: 1-10), we are re-using them across, which also helps in maintenance at our end.

Full dev apis build is not in our near term scope.

Regards,
Vinay

AndreyButokNXP, 2020-03-02:
> "helps in maintenance at our end." 

It's OK..., but not a way when it limits usability.
We have to run all PSA Dev. API test-suites in one project. In current state, it's not possible.

kotegowder, 2020-03-03:
Thanks for the feedback.
We will take this input for the future release.

AndreyButokNXP, 2020-03-03:
Thank you for accepting the improvement request.

yanpengfei66ok, 2020-03-05:
Here's my question:

I need your help，thank you!

After Installing several softwares for building the compilation environment according to the Software Requirements Document which is in the "https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/sw_requirements.md", when I compiled the source code psa-arch-tests of PSA API compliance test suite, the cygwin prompted the following error result:

luofeng66ok@DESKTOP ~/TFMB/psa-arch-tests/api-tests/build
$ cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=ARMCLANG -DTARGET=tgt_dev_apis_tfm_an521 -DCPU_ARCH=armv8m_ml -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS="/cygdrive/c/cygwin64/home/anpengfei/TFMB/trusted-firmware-m/interface/include/psa/crypto.h"
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/cygwin64/home/luofeng66ok/TFMB/psa-arch-tests/api-tests/BUILD/CMakeF
iles/CMakeOutput.log".
See also "C:/cygwin64/home/luofeng66ok/TFMB/psa-arch-tests/api-tests/BUILD/CMakeF
iles/CMakeError.log".

the "CMakeError.log" displays:
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/cygwin64/bin/cc 
Build flags: 
Id flags:  

The output was:
拒绝访问。

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: C:/cygwin64/bin/cc 
Build flags: 
Id flags: -c 

The output was:
拒绝访问。
......
my installed softwares are:

Host Operating System : Windows 10 64bit

Scripting tools :

luofeng66ok@DESKTOP-STB70QG ~/TFMB/psa-arch-tests/api-tests/build

$ Python --version

Python 3.7.0

Cross Compiler toolchain :

luofeng66ok@DESKTOP-STB70QG ~/TFMB/psa-arch-tests/api-tests/build

$ armclang --version

Product: MDK Plus 5.29

Component: ARM Compiler 6.13.1

Tool: armclang [5d895d00]

Target: unspecified-arm-none-unspecified

Build tools :

luofeng66ok@DESKTOP-STB70QG ~/TFMB/psa-arch-tests/api-tests/build

$ cmake --version

cmake version 3.10.0-rc1

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ cygcheck --version

cygcheck (cygwin) 3.1.4

and I have imported Arm Compiler in my environment variable,

enter image description here

at the same time , I also imported Arm Compiler by this way:

C:\cygwin64\home\luofeng66ok.bashrc

export PATH=/cygdrive/c/Keil/ARM/ARMCLANG/bin:$PATH
export PATH=/cygdrive/c/cmake/bin:$PATH
export PATH=/cygdrive/c/armgcc/bin:$PATH
my steps for building psa-arch-tests are:

 git clone https://github.com/ARM-software/psa-arch-tests.git

 git clone https://github.com/laurencelundblade/QCBOR.git (I do not why do this)

 cd <TF-M base folder>
 git clone https://git.trustedfirmware.org/trusted-firmware-m.git
 git clone https://github.com/ARMmbed/mbed-crypto.git -b mbedcrypto-3.0.1
 git clone https://github.com/ARM-software/CMSIS_5.git -b 5.5.0
 git-lfs

my dir tfmb is

luofeng66ok@DESKTOP-STB70QG ~/tfmb
$ ls
CMSIS_5  mbed-crypto  psa-arch-tests  trusted-firmware-m

Start compiling psa-arch-tests

cd api-tests
    mkdir BUILD
    cd  BUILD
    cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=ARMCLANG -DTARGET=tgt_dev_apis_tfm_an521 -DCPU_ARCH=armv8m_ml -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS="/cygdrive/c/cygwin64/home/anpengfei/TFMB/trusted-firmware-m/interface/include/psa/crypto.h"    
as the errors above shows in my cygwin, I do not why?

luofeng66ok@DESKTOP-STB70QG ~/TFMB/psa-arch-tests/api-tests/build
$ cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=ARMCLANG -DTARGET=tgt_dev_apis_tfm_an521 -DCPU_ARCH=armv8m_ml -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS="/cygdrive/c/cygwin64/home/anpengfei/TFMB/trusted-firmware-m/interface/include/psa/crypto.h"
-- The CXX compiler identification is unknown
CMake Error in CMakeLists.txt:
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "C:/cygwin64/home/luofeng66ok/TFMB/psa-arch-tests/api-tests/BUILD/CMakeF
iles/CMakeOutput.log".
See also "C:/cygwin64/home/luofeng66ok/TFMB/psa-arch-tests/api-tests/BUILD/CMakeF
iles/CMakeError.log".

AndreyButokNXP, 2020-03-06:
Hi yanpengfei66ok, your comment is related to the main issue.
Please create a new one.

kotegowder, 2020-04-27:
Hi Andrej,

We have done some work on Storage build requirements.
The changes are available as PR (https://github.com/ARM-software/psa-arch-tests/pull/184)

Please do review and let us know are there any build or functional issues.

kotegowder, 2020-05-06:
Hi Andrej,

The master is having changes that enable for combined ITS and PS build.
Please do check and raise issue if there are any concerns.

Closing the issue for now.

closed 2020-05-06 by kotegowder

AndreyButokNXP, 2020-05-06:
Thank you.
 I will check it next month (hardly occupied by other tasks now) and let you know if any issue.

AndreyButokNXP, 2020-06-09:
Hi,
Thank you very much.  So now, it's possible compile and run ITS and PS test suites in one project.

Minor suggestion:
You have added new STORAGE configuration definition.
But it's absolutely not  needed. It's enough to use existing INTERNAL_TRUSTED_STORAGE and PROTECTED_STORAGE.
So instead of:
#if defined(STORAGE) || defined(INTERNAL_TRUSTED_STORAGE) || defined(PROTECTED_STORAGE)
Better to use:
#if defined(INTERNAL_TRUSTED_STORAGE) || defined(PROTECTED_STORAGE)

kotegowder, 2020-06-09:
It is added to give a user a feel of single option(STORAGE), which presumably covering both ITS and PS.

AndreyButokNXP, 2020-06-09:
Yes, it's clear. 
But this STORAGE option is redundant - it's possible to make absolutely the same configuration without it. Also, if defined STORAGE , no sense if defined or not defined INTERNAL_TRUSTED_STORAGE or/and PROTECTED_STORAGE. It's a bit strange.
It's not critical, just to make the design more clear. It's up to you.