ARM-software/psa-arch-tests #151: MbedCrypto PSA test fail

AndreyButokNXP, 2020-03-02:
Please look at the mbedCrypto issue: https://github.com/ARMmbed/mbed-crypto/issues/380

So, the mbedCrypto projects need clarification for for the failed PSA Test-Suite tests:
- “psa_asymmetric_encrypt does not have support for ECC keys” — that's true, the specification currently does not define any algorithm for psa_asymmetric_encrypt that uses ECC keys. What's the problem there?
- For the incorrect key derivation error codes, what are the problematic inputs?
- For “psa_generate_key generates incorrect key length for RSA”, what are the problematic inputs?

jaypit02, 2020-03-03:
Thanks Andrey for notify this.  Please find my inputs:

> “psa_asymmetric_encrypt does not have support for ECC keys” — that's true, the specification currently does not define any algorithm for psa_asymmetric_encrypt that uses ECC keys. What's the problem there?
Architecturally this is allowed. Hence test will be updated soon to relax this check if psa_asymmetric_encrypt does not have support for ECC keys. 

> For` the incorrect key derivation error codes, what are the problematic inputs?
There is a known issue and it is being discussed with mbed-crypto team here : #175

> For “psa_generate_key generates incorrect key length for RSA”, what are the problematic inputs?
This is a test issue and test will be updated soon. Related issue - https://github.com/ARM-software/psa-arch-tests/issues/155

Also the failure analysis document will be updated soon to point out Github issue links where these are being discussed.

gowthamsiddarthd, 2020-03-03:
Hi @AndreyButokNXP,

>“psa_asymmetric_encrypt does not have support for ECC keys” — that's true, the specification currently does not define any algorithm for psa_asymmetric_encrypt that uses ECC keys. What's the problem there?

>For “psa_generate_key generates incorrect key length for RSA”, what are the problematic inputs?

The PSA test suite has been updated with the changes. - https://github.com/ARM-software/psa-arch-tests/pull/156

closed 2020-03-03 by gowthamsiddarthd