ARM-software/psa-arch-tests #153: docs: PSA ACK tests should have more details on PSA crypto failures

soby-mathew, 2020-03-02:
Hi,
Currently the failures in PSA ACK tests are documented in this page : https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/test_failure_analysis.md

But it does not tell which component is to blame and where the fix has to be applied. Ideally, there has to be an issue corresponding to each failure in the respective project (mbed-crypto, psa-arch-tests, or TF-M). The issue also has to be the ACKed by the project owner as to the reason being mentioned. This is necessary for consumers of the ACK tests to understand the cause of the failure and be able to make informed decisions PSA-crypto integration.

soby-mathew, 2020-03-02:
Please see some discussion here : https://lists.trustedfirmware.org/pipermail/tf-m/2020-March/000795.html

soby-mathew, 2020-03-02:
This is a related issue : https://github.com/ARM-software/psa-arch-tests/issues/151

jaypit02, 2020-03-03:
Thanks Soby for notifying this.

The failure analysis document will be updated soon to point out github issue links where these failure discussions are happening.

gowthamsiddarthd, 2020-03-03:
Documentation has been updated in the PR: https://github.com/ARM-software/psa-arch-tests/pull/156

closed 2020-03-03 by gowthamsiddarthd