ARM-software/psa-arch-tests #154: TEST: 243 is failed

AndreyButokNXP, 2020-03-02:
The latest TFM is failed on TEST: 243, but this information is not present in the list of known fails:
https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/test_failure_analysis.md

TFM returns the error code (-130), but the PSA Test Suite expects (-135).

******************************************
TEST: 243 | DESCRIPTION: Testing crypto key derivation APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_raw_key_agreement - ECDH SECP256R1
[Check 2] Test psa_raw_key_agreement - Small buffer size
[Check 3] Test psa_raw_key_agreement - Invalid usage
[Check 4] Test psa_raw_key_agreement - Unknown KDF
[Check 5] Test psa_raw_key_agreement - Not a key agreement alg
[Check 6] Test psa_raw_key_agreement - Public key on different curve
        Failed at Checkpoint: 4
        Actual: -130
        Expected: -135
TEST RESULT: FAILED (Error Code=0x1)
******************************************

gowthamsiddarthd, 2020-03-03:
Hi @AndreyButokNXP ,

Test is doing negative check and expects error code for the invalid inputs. Since input arguments validation can be in any order, the return code could be different if the order change. Somehow the return value matches with expected value for library mode but not for IPC. Fixed this issue as part of PR: https://github.com/ARM-software/psa-arch-tests/pull/156

Regards,
Gowtham

closed 2020-03-03 by gowthamsiddarthd

AndreyButokNXP, 2020-03-03:
Thank you very much!