ARM-software/psa-arch-tests #162: 64-bit build support for psa-arch-tests required

jainvikas8, 2020-03-17:
Recently when using MinGW64 on Windows system came across an error that the psa-arch-tests can't be built.

--- Config ---
`cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=GNUARM -DTARGET=tgt_dev_apis_tfm_musca_a -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS=`cygpath -am ../../../mbed-crypto/include` -DCPU_ARCH=armv8m_ml
cmake --build . --  -j`

--- Error --
`C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib/libmingwex.a when searching for -lmingwex
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib\libmingwex.a when searching for -lmingwex
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib/libmingwex.a when searching for -lmingwex
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lmingwex
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib/libmsvcrt.a when searching for -lmsvcrt
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib\libmsvcrt.a when searching for -lmsvcrt
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: skipping incompatible C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/lib/libmsvcrt.a when searching for -lmsvcrt
C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/8.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lmsvcrt
collect2.exe: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/TargetConfigGen.dir/build.make:98: TargetConfigGen.exe] Error 1
make[4]: *** [CMakeFiles/Makefile2:68: CMakeFiles/TargetConfigGen.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make.exe[2]: *** [CMakeFiles/psa_generate_database.dir/build.make:75: src/psa_generate_database-stamp/psa_generate_database-install] Error 2
make.exe[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/psa_generate_database.dir/all] Error 2
make.exe: *** [Makefile:84: all] Error 2`

Using - https://github.com/ARMmbed/mbed-crypto/tree/mbedcrypto-3.0.1 which supports 64-bit build.

I see compiler flags set during the build:
`# set_target_properties(${PROJECT_NAME} PROPERTIES COMPILE_FLAGS "-m32 -w")`
`# set_target_properties(${PROJECT_NAME} PROPERTIES LINK_FLAGS "-m32")`

So, I think there is a requirement to support 64-bit support.

Patater, 2020-03-17:
This is 64-bit host support, right? Supporting building the psa-arch-tests on a 64-bit build host, targeting a 32-bit ARMv8-M.

jainvikas8, 2020-03-17:
> This is 64-bit host support, right? Supporting building the psa-arch-tests on a 64-bit build host, targeting a 32-bit ARMv8-M.

yes.

jainvikas8, 2020-03-17:
This is possible duplication of this https://github.com/ARM-software/psa-arch-tests/issues/119

kotegowder, 2020-05-18:
We do support building psa-arch-tests on 64-bit host but it generates 32-bit archives for v7m and v8m targets. There is also an option to build for host machine itself with -DTOOLCHAIN=HOST_GCC, in which case the archive is for host machine.

jainvikas8, 2020-05-18:
@kotegowder 
Thanks for the reply.

But I still seem to get this error on my linux system
`Scanning dependencies of target psa_generate_database_prerequisite
[  0%] Built target psa_generate_database_prerequisite
Scanning dependencies of target psa_generate_database
[  1%] Creating directories for 'psa_generate_database'
[  3%] No download step for 'psa_generate_database'
[  4%] No patch step for 'psa_generate_database'
[  6%] No update step for 'psa_generate_database'
[  8%] Performing configure step for 'psa_generate_database'
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Work/Workspace/tf-m/psa-arch-tests/api-tests/BUILD/src/psa_generate_database-build
[  9%] No build step for 'psa_generate_database'
[ 11%] Performing install step for 'psa_generate_database'
[ 20%] [PSA] : Creating generator source /mnt/c/Work/Workspace/tf-m/psa-arch-tests/api-tests/BUILD/targetConfigGen.c
Scanning dependencies of target TargetConfigGen
[ 40%] Building C object CMakeFiles/TargetConfigGen.dir/mnt/c/Work/Workspace/tf-m/psa-arch-tests/api-tests/BUILD/targetConfigGen.c.o
[ 60%] Linking C executable TargetConfigGen
[ 60%] Built target TargetConfigGen
Scanning dependencies of target run_generator_output
[ 80%] [PSA] : Creating output header target_database.h
./TargetConfigGen: 1: ./TargetConfigGen: Syntax error: word unexpected (expecting ")")
CMakeFiles/run_generator_output.dir/build.make:77: recipe for target 'target_database.h' failed
make[5]: *** [target_database.h] Error 2
CMakeFiles/Makefile2:97: recipe for target 'CMakeFiles/run_generator_output.dir/all' failed
make[4]: *** [CMakeFiles/run_generator_output.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/psa_generate_database.dir/build.make:94: recipe for target 'src/psa_generate_database-stamp/psa_generate_database-install' failed
make[2]: *** [src/psa_generate_database-stamp/psa_generate_database-install] Error 2
CMakeFiles/Makefile2:184: recipe for target 'CMakeFiles/psa_generate_database.dir/all' failed
make[1]: *** [CMakeFiles/psa_generate_database.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2`

for

`cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=GNUARM -DTARGET=tgt_dev_apis_tfm_musca_a -DSUITE=STORAGE -DCPU_ARCH=armv8m_ml -DPSA_INCLUDE_PATHS=readlink -f ../../../trusted-firmware-m/interface/include`

on x64 system.

If i disable 
`# set_target_properties(${PROJECT_NAME} PROPERTIES COMPILE_FLAGS "-m32 -w")`
`# set_target_properties(${PROJECT_NAME} PROPERTIES LINK_FLAGS "-m32")`
then everything works fine.

kotegowder, 2020-05-19:
Please go through the SW requirements for psa-arch-tests.
Available at https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/sw_requirements.md.

You need to have the right version of the tools mentioned and it should support 32-bit binary generation.

jainvikas8, 2020-05-19:
The tools are 32-bit for `Host Compiler toolchain: GCC 5.4.0 32-Bit (Linux Host) or MinGW 6.3.0 32-Bit (Windows Host)`

That's the point, there needs to be 64-bit support for this?

kotegowder, 2020-05-19:
Can you please switch to Host GCC 5.4.0 and give it a try?
Let us know if you still see the the build issue.

kotegowder, 2020-05-19:
I see you are trying to build for Musca A.
At present our design require support of host compiler which supports intermediate 32-bit executable generation which in turn

Patater, 2020-05-19:
There seems to be some confusion about the difference between host and target here.
- Host: the system you are building on. Can run 64-bit compilers that target 32-bit targets.
- Target: the system you are building for. Can run 32-bit binaries.

Vikas can correct me if I'm wrong, but I believe he's looking to target a 32-bit platform from his 64-bit host. It's this arrangement which is currently broken at the moment, hence his feature request for the ability to use 64-bit host compilers to build for 32-bit targets.

kotegowder, 2020-05-20:
The existing build step of psa-arch-tests involves generating intermediate executable which is trying to output a table (32-bit target compliant) and is used by our infrastructure code during run-time.

This requires re-look at the build design at our end. We we shall analyse and see how we can scale across.

closed 2020-08-13 by kotegowder