ARM-software/psa-arch-tests #163: Test psa_generate_key with RSA 2048 Keypair is enabled even if RSA is disabled

AndreyButokNXP, 2020-03-19:
Hello,

There is an issue, when RSA is disabled in api-tests/platform/targets/[board]/crypto/pal_crypto_config.h:
```
//#define ARCH_TEST_RSA
// #define ARCH_TEST_RSA_1024
// #define ARCH_TEST_RSA_2048
// #define ARCH_TEST_RSA_3072
```

the  RSA 2048 Keypair is still run:
TEST: 216 | DESCRIPTION: Testing crypto generator functions APIs
..
[Check 4] Test psa_generate_key with RSA 2048 Keypair
...

It is causing the failed test for a platform with disabled RSA:
```
TEST: 216 | DESCRIPTION: Testing crypto generator functions APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_generate_key 16 Byte AES
[Check 2] Test psa_generate_key 24 Byte AES
[Check 3] Test psa_generate_key 32 Byte AES
[Check 4] Test psa_generate_key with RSA 2048 Keypair
        Failed at Checkpoint: 3
        Actual: -134
        Expected: 0
TEST RESULT: FAILED (Error Code=0x1)
```

Guess, the check 4 must be skipped when RSA is disabled.

Thanks,
Andrej Butok

AndreyButokNXP, 2020-03-19:
The proposed fix is to add missing:
```
#ifdef ARCH_TEST_RSA
#ifdef ARCH_TEST_RSA_2048
```
to psa-arch-tests\api-tests\dev_apis\crypto\test_c016\test_data.h line 78:

```
#ifdef ARCH_TEST_RSA
#ifdef ARCH_TEST_RSA_2048
{"Test psa_generate_key with RSA 2048 Keypair\n", 7, PSA_KEY_TYPE_RSA_KEY_PAIR,
 PSA_KEY_USAGE_EXPORT | PSA_KEY_USAGE_SIGN | PSA_KEY_USAGE_VERIFY, PSA_ALG_RSA_PKCS1V15_SIGN_RAW,
 2048, 2048, {1190, 1194}, PSA_SUCCESS
},
#endif
#endif
```

Is it OK?

kotegowder, 2020-03-23:
Hi Andrey,

There are other dependent macros which require RSA. 
Like

ARCH_TEST_RSA_PKCS1V15_SIGN
ARCH_TEST_RSA_PKCS1V15_SIGN_RAW
ARCH_TEST_RSA_PKCS1V15_CRYPT

I assume there are too commented when you commented 

//#define ARCH_TEST_RSA
//#define ARCH_TEST_RSA_1024
//#define ARCH_TEST_RSA_2048
//#define ARCH_TEST_RSA_3072

And yes the changes proposed are required and will consider them.

Thank you for the finding.

AndreyButokNXP, 2020-03-23:
Yes. I have disabled this set of RSA definitions too.

kotegowder, 2020-03-27:
Closing the issue as fix made available on master.

closed 2020-03-27 by kotegowder