ARM-software/psa-arch-tests #164: Print garbage message on MPS2 AN521 board

edison-ai, 2020-03-23:
When running PSA Arch Dev API test on MPS2 AN521 board, the TF-M boot message works normally. But the PSA Arch test will print out garbage messages.

Please confirm this and try to fix that if it exists.

ChangwuShan, 2020-05-15:
I have confirmed in my side and this diff can eliminate the garbage messages issue.
--- a/api-tests/platform/drivers/uart/cmsdk/pal_uart.c
+++ b/api-tests/platform/drivers/uart/cmsdk/pal_uart.c
@@ -31,7 +31,7 @@ void pal_uart_cmsdk_init(uint32_t uart_base_addr)
     ((uart_t *) g_uart)->CTRL = CMSDK_UART_CTRL_TXEN_Msk;
     /* However, the UART resets to a BAUDDIV of 0, which is invalid. Set it to
     * the minimum valid number. */
-    ((uart_t *) g_uart)->BAUDDIV = 16;
 }

kotegowder, 2020-05-15:
The changes are available in PR
https://github.com/ARM-software/psa-arch-tests/pull/202

kotegowder, 2020-05-15:
Please review and confirm

TTornblom, 2020-05-15:
Yes, this seems to fix the UART issue.

Thanks

kotegowder, 2020-05-15:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/202

kotegowder, 2020-05-15:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/202
Closing the issue.

closed 2020-05-15 by kotegowder