ARM-software/psa-arch-tests #165: Source of a RAM wasting

AndreyButokNXP, 2020-03-23:
Hello,
Most of `test_data`, that are defined in _test_data.h_, contain read-only data, but they are defined as RW `static test_data`. This is the source of huge wasting of RAM.
After the fix we have saved >20KB of RAM. So it's also good to have it in the original project.
The fix:
1) Most of  `static test_data` in test_data.h must be defined as  `static const test_data`
   Exception is for test_a001,c025,c039,c040,c050.
2) To avoid warnings, change `char *string` to `cons char *string` in `print` functions:
     - ` val_status_t val_print(print_verbosity_t verbosity, char *string, int32_t data); => val_status_t val_print(print_verbosity_t verbosity, const char *string, int32_t data);`
     - `val_status_t (*print) (print_verbosity_t verbosity, char *string, int32_t data); => val_status_t (*print) (print_verbosity_t verbosity, const char *string, int32_t data);`
     - `int pal_print_ns(char *str, int32_t data); => int pal_print_ns(const char *str, int32_t data);`

Hope, you are OK with it.

AndreyButokNXP, 2020-03-23:
Additional testing has showed that putting it to `const` is causing fail. 
So closing the issue so far.

AndreyButokNXP, 2020-03-24:
Reopening issue. 
The previous comments are correct. In addition to avoid writing to ROM memory, it is needed to use local variables instead of fields. Mainly for `key_handle`
For example:
```
psa_key_handle_t        key_handle = check1[i].key_handle; 
...
status = val->crypto_function(VAL_CRYPTO_IMPORT_KEY, &attributes, key_data,
                 check1[i].key_length, &key_handle);
```
Saving >20KB is a good argument.

AndreyButokNXP, 2020-04-02:
One more source of savings:
`client_test_t` change to `const client_test_t`. 
It will save >0.5KB of RAM

AndreyButokNXP, 2020-04-03:
Another source of savings:
Many cryptographic tests have own static buffers, like: `static uint8_t  data[BUFFER_SIZE];`
Putting them to the stack (inside function scope, without `static`) is saving >10KB of RAM.

kotegowder, 2020-04-03:
Hi Andrey,

Your inputs are vital.
Could you please upstream changes as PR ? if possible.
We shall take a look and the changes.

AndreyButokNXP, 2020-04-06:
>Could you please upstream changes as PR ? if possible.

Most probably, I may do it only somewhere in Q3/4.

kotegowder, 2020-04-09:
We are doing some changes based on your comments, will be up-streaming them to master in a week or so.

kotegowder, 2020-04-27:
Hi Andrej,

We have done some changes mainly with the use of 'const' qualifier wherever it is necessary based on your feedback.
The changes are available at (https://github.com/ARM-software/psa-arch-tests/pull/184)

Please do review and let us know for any build or functional issues.

kotegowder, 2020-05-06:
Hi Andrej,

The master has changes that addresses some of the enhancements you pointed out.
Please do check at your end.

Closing the issue for now.

closed 2020-05-06 by kotegowder

AndreyButokNXP, 2020-05-06:
Thank you.
I will check it next month (hardly occupied by other tasks now) and let you know if any issue.

AndreyButokNXP, 2020-06-09:
Hi, 
Thank you again. The all  added enchantments are correct.

Let's continue with improvements:
1) Change `test_data`  to `const test_data`
for the rest of the tests: test_c024, test_c025, test_c039, test_c040 , test_c050
2) Put all global static buffers to the scope of of functions (on stack) => saves huge amount of RAM
For example:
```
//OLD - static uint8_t     data[BUFFER_SIZE];
int32_t psa_mac_abort_test(caller_security_t caller)
{
    uint8_t     data[BUFFER_SIZE]; //NEW
```
For the tests: test_c003, test_c004, test_c016, test_c017,   test_c020,  test_c022, test_c027, test_c028, test_c031, test_c036, test_c037, test_c038, test_c039, test_c040, test_c041, test_c043, test_c044, test_c046, test_c048, test_c049, test_c050, test_c059, test_c061

If any issue or difficulties with the changes, write me I will help.

Thank you,
Andrej Butok

kotegowder, 2020-06-09:
Hi Andrej,

I am re-working on re structuring the test data used across crypto tests which possibly reduce footprint.

Will keep you update once the changes reflect in master, so that you can take a look at it and give us valuable feedback.

kotegowder, 2020-07-23:
Hi Andrej,

We have re-worked Crypto tests(most of) and the changes are available now in master.
Let us know if there is improvement in terms of reduced RAM usage. 

Some of the tests (left over) shall be re-worked as part of next release.

Once again, happy to receive feedback..

Thank you,
Vinay

AndreyButokNXP, 2020-07-23:
Hi Vinay,
I have just updated to the latest version of the Test Suite. See the big step forward in optimization. 

But the Crypto Suite is failed (before it was OK). Is this known issue?

```
************ Crypto Suite Report **********
TOTAL TESTS     : 40
TOTAL PASSED    : 27
TOTAL SIM ERROR : 0
TOTAL FAILED    : 11
TOTAL SKIPPED   : 2
******************************************
```
The failed tests:

```
******************************************

TEST: 224 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_encrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 225 | DESCRIPTION: Testing crypto AEAD APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_aead_decrypt - CCM - AES - 13B nonce & 8B add data
        Failed at Checkpoint: 9
        Actual: -136
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 236 | DESCRIPTION: Testing crypto symmetric cipher APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_cipher_update - Encrypt - AES CBC_NO_PADDING
[Check 2] Test psa_cipher_update - Encrypt - AES CBC_NO_PADDING (Short in)
[Check 3] Test psa_cipher_update - Encrypt - AES CTR
[Check 4] Test psa_cipher_update - Encrypt - small output buffer size
[Check 5] Test psa_cipher_update - Decrypt - AES CBC_NO_PADDING
[Check 6] Test psa_cipher_update - Decrypt - AES CBC_NO_PADDING (Short in)
[Check 7] Test psa_cipher_update - Decrypt - AES CTR
[Check 8] Test psa_cipher_update - Decrypt - small output buffer size
[Check 9] Test psa_cipher_update without cipher setup
        Failed at Checkpoint: 2
        Actual: -130
        Expected: -137

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 237 | DESCRIPTION: Testing crypto symmetric cipher APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_cipher_finish - Encrypt - AES CBC_NO_PADDING
[Check 2] Test psa_cipher_finish - Encrypt - AES CBC_NO_PADDING (Short in)
[Check 3] Test psa_cipher_finish - Encrypt - AES CTR
[Check 4] Test psa_cipher_finish - Encrypt - AES CTR (short input)
[Check 5] Test psa_cipher_finish - Decrypt - AES CBC_NO_PADDING
[Check 6] Test psa_cipher_finish - Decrypt - AES CBC_NO_PADDING (Short in)
        Failed at Checkpoint: 7
        Actual: -130
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)

******************************************

TEST: 239 | DESCRIPTION: Testing crypto asymmetric APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_asymmetric_encrypt - Invalid algorithm
        Failed at Checkpoint: 4
        Actual: -130
        Expected: -135

TEST RESULT: FAILED (Error Code=0x1)

```
I am using the latest TFM. Any suggestion?

kotegowder, 2020-07-23:
Hi Andrej,

I used latest psa-arch-tests, TF-Mv1.1-RC2 and AN521 target.
I get below result:

************ Crypto Suite Report **********
TESTS     : 61
TOTAL PASSED    : 39
TOTAL SIM ERROR : 0
TOTAL FAILED    : 21
TOTAL SKIPPED   : 1
******************************************

Our next release will have pretty good coverage with Crypto 1.0.0, however it is going to take sometime.
Keep you update on that.

AndreyButokNXP, 2020-07-24:
Probably, I am using newer TF-M v1.1-RC2, released 2020-07-03 (aacfe93b15e5a7ee51d54ef47e6dd1dfe9c65d27)

I have found sources of the issues, one of them for test_c024.c

```
        status = val->crypto_function(VAL_CRYPTO_IMPORT_KEY,
                                      &attributes,
                                      check1[i].data,
                                      check1[i].data_length,
                                      &handle);
        TEST_ASSERT_EQUAL(status, PSA_ERROR_NOT_SUPPORTED, TEST_CHECKPOINT_NUM(8));

       status = val->crypto_function(VAL_CRYPTO_DESTROY_KEY, handle);
       TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(9));
```

The check 9 is always failed, because `handle` is set to Zero (not valid handle) after VAL_CRYPTO_IMPORT_KEY finished with PSA_ERROR_NOT_SUPPORTED

So tests are not able to pass with this behavior.

I am going to fix tests one by one, are you interested in the result/changes?

kotegowder, 2020-07-24:
Hi Andrej,

Yes, ideally checkpoint 8 and 9 need to swapped. The result is same as yours with the TF-M tag as specified earlier.
Since I tested with latest mbedtls directly even the existing order worked. As the latest crypto (1.0.0) says psa_crypto_destroy will always return SUCCESS for PSA_KEY_ID_NULL(0).

I will check for these order of checks and push PR. 
Will let you know when I make it available.