ARM-software/psa-arch-tests #166: RAW Keys for RSA Tests

zaya-mc, 2020-03-23:
Hi all, 

We are trying to implement PSA Functional API for a custom environment. We don't use mbed-crypto/mbedTLS.
The secure element supports RSA but accepts only RAW Keys (1024, 2048, 4096 bits etc) but PSA Arch Tests test using DER encoded keys.
Is that somehow possible to add a compiler switch to support both DER and RAW keys? 
Otherwise, we have to disable all RSA tests which dramatically reduces the coverage of PSA Functional API Certification.

I don't see a requirement like encoded keys in the PSA L1 Questionaire also. 

Thoughts? 

Thank you.
Murat

gilles-peskine-arm, 2020-03-25:
Hello,

I can't speak about the test suite but I can speak from the point of view of the specification.

A compliant implementation must use the standard representation (defined in RFC 8017 and RFC 3279 for RSA private/public keys). All implementations must support the same representation to allow interoperability. See the documentation of `psa_import_key`, `psa_export_key` and `psa_export_public_key`.

Most hardware accelerators and secure elements don't use this representation, of course. There are many different “raw” representations, with different alignment, ordering of fields, packing, endianness, etc. It's up to the driver for the hardware to translate between the import/export format and the internal representation.

If your implementation does not support the import or export of private keys in production, you may need to implement it to pass the test suite? I'm not sure about that part.

athoelke, 2020-03-25:
**psa-arch-tests** is following the PSA Cryptography API specification, which expects that an implementation of `psa_import_key()` will support the DER encoded format for an RSA key, in addition to any implementation-specific data formats.

In the description of `psa_import_key()` - (my bold):
> `data`
>
> Buffer containing the key data. The content of this buffer is interpreted according to the type declared in `attributes`. **All implementations must support at least the format described** in the documentation of `psa_export_key()` or `psa_export_public_key()` for the chosen type. **Implementations can support other formats**, but be conservative in interpreting the key data: it is recommended that implementations reject content if it might be erroneous, for example, if it is the wrong type or is truncated.

The relevant section in `psa_export_public_key()` says:
> For standard key types, the output format is as follows:
>
> *  For RSA public keys, with key type `PSA_KEY_TYPE_RSA_PUBLIC_KEY`, the DER encoding of the representation defined by _Algorithms and Identifiers for the Internet X.509 Public Key Infrastructure Certificate and Certificate Revocation List (CRL) Profile_ RFC3279 §2.3.1 as `RSAPublicKey`.
>
>    ```
>    RSAPublicKey ::= SEQUENCE {
>       modulus            INTEGER,    -- n
>       publicExponent     INTEGER  }  -- e
>    ```

How important is it that your API uses RAW format for import of keys?

Unambiguously separating a DER-encoded key and a RAW key during import might depend on certainty that the DER-encoding is always at least one octet longer that the length of the RAW key?

athoelke, 2020-03-25:
@gilles-peskine-arm - you beat me to it. 😄

kotegowder, 2020-04-02:
Hi Murat,

psa-arch-tests test the implementation against the spec recommendation and agree with the answers from Gilles and Andrew.

Do let us know your feedback. If all OK, we can close the issue.

zaya-mc, 2020-04-03:
Hi all, thank you for the answers. That's fine. We can close the ticket. Thank you.

closed 2020-04-03 by zaya-mc