ARM-software/psa-arch-tests #170: Split dev-api-tests.

jainvikas8, 2020-03-30:
The idea is to split dev-api-tests or modularise them for targets like MUSCA_A1 which have limited resources to run them.

For example:
---PSA API tests config---
```
cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=GNUARM -DTARGET=tgt_dev_apis_tfm_musca_a -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS=`readlink -f ../../mbed-crypto/include` -DCPU_ARCH=armv8m_ml
```
when used with TF-M
---TF-M build config---
```
cmake -G"Unix Makefiles" -DPROJ_CONFIG=`readlink -f ../configs/ConfigPsaApiTest.cmake` -DPSA_API_TEST_CRYPTO=ON -DTARGET_PLATFORM=MUSCA_A -DCOMPILER=GNUARM ../
```
results in 
```[ 83%] Linking C executable tfm_ns.axf
Memory region         Used Size  Region Size  %age Used
           FLASH:      176808 B       382 KB     45.20%
             RAM:      127072 B        32 KB    387.79%
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: tfm_ns.axf section `.data' will not fit in region `RAM'
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 94304 bytes
collect2: error: ld returned 1 exit status
app/CMakeFiles/tfm_ns.dir/build.make:127: recipe for target 'app/tfm_ns.axf' failed
make[2]: *** [app/tfm_ns.axf] Error 1
CMakeFiles/Makefile2:1271: recipe for target 'app/CMakeFiles/tfm_ns.dir/all' failed
make[1]: *** [app/CMakeFiles/tfm_ns.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
possibly due to https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/dev_apis/crypto/testsuite.db number of tests.

So it would be better to split these tests in modules or config option like `TEST=PART1..n` or `TEST=kind-of-tests` 

This focuses on user experience and would benefit future targets using these tests.

kotegowder, 2020-04-03:
There is new pull request available which address this enhancement.

https://github.com/ARM-software/psa-arch-tests/pull/171

Can you please try at your end and let us know?

AndreyButokNXP, 2020-04-03:
The split will not needed if to apply all proposed improvements: https://github.com/ARM-software/psa-arch-tests/issues/165

kotegowder, 2020-04-03:
Hi Andrey,

We have a plan to rework on the usage of global and using const wherever appropriate.
Your inputs are vital from other issues you created.  

This new CMake configuration option also enable to debug single or selective tests without actually having all of them in binary.

kotegowder, 2020-04-07:
The support is provided under master.
Closing the issue.

closed 2020-04-07 by kotegowder