ARM-software/psa-arch-tests #174: Wrong Hash Size in Crypto 6 Tests

zaya-mc, 2020-04-09:
Hello, 

In the Crypto Test 6, it passes hash_size values as hash buffer size (sizeof(hash)) and constant for all tests cases. 
But it fails for "Small Buffer Size" test case because it does not pass the test case hash size value.

```
/* Calculate the hash (digest) of a message */
status = val->crypto_function(VAL_CRYPTO_HASH_COMPUTE, check1[i].alg, &check1[i].input,
                 check1[i].input_length, hash, **hash_size**, &hash_length);
TEST_ASSERT_EQUAL(status, check1[i].expected_status, TEST_CHECKPOINT_NUM(3));
```

I think we need to pass check1[i].hash_length which passes small length size and passes the tests. 
If you are agree, I can create a pull request for it.

kotegowder, 2020-04-13:
Hello,

Can you please create PR? 
We shall review them.

Thank you.

kotegowder, 2020-04-20:
Hi Murat,

Let us know, if you have plan to raise PR for the below issues.

https://github.com/ARM-software/psa-arch-tests/issues/174
https://github.com/ARM-software/psa-arch-tests/issues/175
https://github.com/ARM-software/psa-arch-tests/issues/176

zaya-mc, 2020-05-04:
@kotegowder Apologise, I did not get a notification and was busy with internal works. Sure, I can create pull requests for all these issues to discuss. This week or next week. FYI Regards.
Murat

kotegowder, 2020-05-15:
Addressed with https://github.com/ARM-software/psa-arch-tests/pull/186.
Closing the issue.

closed 2020-05-15 by kotegowder