ARM-software/psa-arch-tests #176: Expected Error for Small Buffer Case in Crypto 42 Tests

zaya-mc, 2020-04-09:
Hi,

There is a test case in Crypto 42, "Small output buffer", I would expect error value should be PSA_ERROR_BUFFER_TOO_SMALL but it is PSA_ERROR_INVALID_SIGNATURE. 

Should it be PSA_ERROR_BUFFER_TOO_SMALL?

kotegowder, 2020-05-15:
Hi Murat,

If you look at the return values for psa_asymmetric_verify at below link
https://github.com/ARMmbed/mbed-crypto/blob/psa-crypto-api-1.0-beta3/include/psa/crypto.h

The more suitable return value for reduced length is PSA_ERROR_INVALID_SIGNATURE.
Since when the implementation tries to verify for the reduced input length the obvious result is PSA_ERROR_INVALID_SIGNATURE.

Which was agreed between psa-arch-tests and TF-M internally. 

Let us know what you think.

kotegowder, 2020-05-19:
@zaya-mc : Do let us know if you have different opinion on this. If you agree, then we can close the issue.

zaya-mc, 2020-05-19:
@kotegowder yes it is absolutely fine, you are right. Let us close the ticket. Thanks.

closed 2020-05-19 by zaya-mc