ARM-software/psa-arch-tests #181: Two level authentication for Attestation Token

zaya-mc, 2020-04-09:
Hi all, 

As an open discussion; due to security reasons, we don't want to sign the attestation using the attestation key because there are some security requirements in our side. 

Instead, we prefer to 
- create an instant key for attestation, 
- sign the instant key using the attestation key in SPE
- sign the attestation token using the instant key
- attach instant key and instant key signature to the attestation token too (to use them during the verification)
Like a chain of trust.

So we can handle this in our side but during the verification, PSA VAL parses the CBOR formatted token in the VAL layer (val_initial_attest_verify_token), which we cannot modify it. 

Is it somehow possible to support two-layer authentication in VAL Layer or can psa-arch-tests leave parsing to the PAL layer?

Thank you
Murat

jk-arm, 2020-12-07:
the custom requirement will not be supported, because psa arch test here to comply with the PSA API specification

closed 2020-12-07 by jk-arm

adrianlshaw, 2020-12-09:
Hi @zaya-mc can you send your requirement to the [PSA Crypto mailing list](https://lists.trustedfirmware.org/mailman/listinfo/psa-crypto) so we can discuss what needs to be added? All the API designers monitor that channel. This feature might be of interest to both crypto and attestation users.