ARM-software/psa-arch-tests #182: Wrong Return Values for Protected Storage in Test 9

zaya-mc, 2020-04-14:
In the tests, the PSA psa_ps_set() called by NULL and 0 data length and it expects success. 

```
    val->print(PRINT_TEST, "[Check 1] Call set API with NULL pointer and data length 0\n", 0);
    status = SST_FUNCTION(s009_data[1].api, uid, 0, NULL, PSA_STORAGE_FLAG_NONE);
    TEST_ASSERT_EQUAL(status, s009_data[1].status, TEST_CHECKPOINT_NUM(1));
```

Herein **s009_data[1].status** value is PSA_SUCCESS;

But psa_ps_set() should return PSA_ERROR_INVALID_ARGUMENT
```
 * \retval PSA_ERROR_INVALID_ARGUMENT      The operation failed because one
 *                                         of the provided pointers(`p_data`)
 *                                         is invalid, for example is `NULL` or
 *                                         references memory the caller cannot
 *                                         access
```

We could modify test_data.h with the proper return codes or we could check for NOT_EQUAL (TEST_ASSERT_NOT_EQUAL)

It is the same for all other test cases.

kotegowder, 2020-04-15:
Hi,

data_length = 0, is considered valid case when UID is valid.

This is being discussed and clarified at : 
https://github.com/ARMmbed/psa_trusted_storage_api/issues/43

Let us know, if that helps?

athoelke, 2020-04-15:
There is an open issue against the specification (https://github.com/ARMmbed/psa_trusted_storage_api/issues/86 - private repo) to clarify the situation regarding valid/invalid data pointers.

The spec is currently inconsistent in the wording used (across the APIs) and ambiguous depending on whether it is read by an application developer or an API implementer.

The intention is that:
* When `data_length == 0`, no data is being provided and the value of `p_data` is ignored.
* When `data_length > 0`, then that many bytes of memory starting at address `p_data` must be readable by the caller.

The API may be present on platforms for which memory location `0` is addressable, so the explicit use of `NULL` in the description of the `PSA_ERROR_INVALID_ARGUMENT` error should be replaced with clearer text. On platforms where `NULL` is never a valid address, then passing `p_data == NULL` with a non-zero length must result in an error or an abort.

kotegowder, 2020-04-20:
The current reference test platform TF-M returns PSA_SUCCESS when data_length = 0 and ignores p_data. 

@zaya-mc : let us know how this is interpreted at your end and changes required ?

kotegowder, 2020-04-27:
Please let us know if there are any gaps, so that we can work and close this together.

zaya-mc, 2020-05-08:
Hi guys, sorry about delayed response. Yes, it is a false alarm, so we can close the ticket. Thank you very much.

closed 2020-05-08 by zaya-mc