ARM-software/psa-arch-tests #183: Verify for AES in Crypto Test 3

zaya-mc, 2020-04-16:
In the Crypto test 3, there is a case "Test psa_export_key with key policy verify", 
in this case key usage type is PSA_KEY_USAGE_VERIFY but the key type is PSA_KEY_TYPE_AES. 

It should be fine for the tests but we are using a Secure Element for key storage and when we pass "Verify" flag for an AES key it returns error which makes sense. 

Can we change the key type to RSA or ECC? 

Thanks
Murat

athoelke, 2020-04-16:
A comment on the current test case.

Although the `PSA_KEY_USAGE_VERIFY` usgae policy can make sense for an AES key - for a CMAC `psa_mac_verify()` operation - it is understandable that some Secure Elements might not implement CMAC operations on AES keys.

I'll leave the engineering team to comment on the possibility of making this configurable or changing to an asymmetric key type.

zaya-mc, 2020-04-23:
@athoelke thank you yes it makes sense for AES CMAC, so let us close the ticket.

closed 2020-04-23 by zaya-mc