ARM-software/psa-arch-tests #185: Wrong Test Data Order in Test 58

zaya-mc, 2020-05-04:
In test 58, 

test data structure is as below; 

```
typedef struct {
    char                    test_desc[75];
    psa_key_handle_t        key_handle;
    psa_key_type_t          key_type;
    uint8_t                 key_data[32];
    uint32_t                key_length;
    psa_key_usage_t         usage;
    psa_algorithm_t         key_alg;
    psa_algorithm_t         alg;
    uint8_t                 plaintext[30];
    uint8_t                 additional_data[32];
    uint8_t                 nonce[32];
    size_t                  nonce_size;
    size_t                  ad_length;
    size_t                  plaintext_length;
    size_t                  input_length;
    size_t                  output_size;
    uint8_t                 expected_output[45];
    size_t                  expected_length;
    psa_status_t            expected_status;
} test_data;
```

But, maybe I am wrong, the nonce and additional data (also length fields) need to be replaced. the nonce length seems 16 while additional data length is 13 for AES CCM. And we use the same 13 bytes nonce in test 24. 

```
{"Test psa_aead_update - AES-CCM\n", 1, PSA_KEY_TYPE_AES,
{0xD7, 0x82, 0x8D, 0x13, 0xB2, 0xB0, 0xBD, 0xC3, 0x25, 0xA7, 0x62, 0x36, 0xDF,
 0x93, 0xCC, 0x6B},
 AES_16B_KEY_SIZE, PSA_KEY_USAGE_ENCRYPT, PSA_ALG_CCM, PSA_ALG_CCM,
{0x45, 0x35, 0xd1, 0x2b, 0x43, 0x77, 0x92, 0x8a, 0x7c, 0x0a, 0x61, 0xc9, 0xf8,
 0x25, 0xa4, 0x86, 0x71, 0xea, 0x05, 0x91, 0x07, 0x48, 0xc8, 0xef},
{0x48, 0xc0, 0x90, 0x69, 0x30, 0x56, 0x1e, 0x0a, 0xb0, 0xef, 0x4c, 0xd9, 0x72},
{0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C,
 0x0D, 0x0E, 0x0F}, 
16,  // nonce_size
13,  // ad_length
24, 24, BUFFER_SIZE,
{0x26, 0xc5, 0x69, 0x61, 0xc0, 0x35, 0xa7, 0xe4, 0x52, 0xcc, 0xe6, 0x1b, 0xc6,
 0xee, 0x22, 0x0d, 0x77, 0xb3, 0xf9, 0x4d, 0x18, 0xfd, 0x10, 0xb6, 0xd8, 0x0e,
 0x8b, 0xf8, 0x0f, 0x4a, 0x46, 0xca, 0xb0, 0x6d, 0x43, 0x13, 0xf0, 0xdb, 0x9b,
 0xe9}, 40,
 PSA_SUCCESS
},
```

zaya-mc, 2020-05-05:
@kotegowder you might be intersted. Thank you.

kotegowder, 2020-05-26:
Changes are pushed as part of https://github.com/ARM-software/psa-arch-tests/pull/193

closed 2020-05-26 by kotegowder