ARM-software/psa-arch-tests #194: Wrong Argument in PSA Crypto 50

zaya-mc, 2020-05-07:
After OpenKey call, it gets all key attributes to "get_attributes" variable and checks all key attributes from the "get_attributes" variable but for the checkpoint 19 and 20, it passes "attributes" instead of "get_attributes" which fails the tests. 

```
            val->crypto_function(VAL_CRYPTO_GET_KEY_USAGE_FLAGS, &attributes, &get_key_usage_flags);
            TEST_ASSERT_EQUAL(get_key_usage_flags, check1[i].usage, TEST_CHECKPOINT_NUM(19));

            val->crypto_function(VAL_CRYPTO_GET_KEY_ALGORITHM, &attributes, &get_key_algorithm);
            TEST_ASSERT_EQUAL(get_key_algorithm, check1[i].key_alg, TEST_CHECKPOINT_NUM(20));

```
needs to be 

```
            val->crypto_function(VAL_CRYPTO_GET_KEY_USAGE_FLAGS, &get_attributes, &get_key_usage_flags);
            TEST_ASSERT_EQUAL(get_key_usage_flags, check1[i].usage, TEST_CHECKPOINT_NUM(19));

            val->crypto_function(VAL_CRYPTO_GET_KEY_ALGORITHM, &get_attributes, &get_key_algorithm);
            TEST_ASSERT_EQUAL(get_key_algorithm, check1[i].key_alg, TEST_CHECKPOINT_NUM(20));

```

kotegowder, 2020-05-07:
Hi Murat,

Regarding issues:
https://github.com/ARM-software/psa-arch-tests/issues/194
https://github.com/ARM-software/psa-arch-tests/issues/195
https://github.com/ARM-software/psa-arch-tests/issues/196

These are all fixed on latest master : https://github.com/ARM-software/psa-arch-tests/tree/master/api-tests

Can you please check and confirm?

zaya-mc, 2020-05-11:
@kotegowder tested. We can close the ticket. Thank you.

closed 2020-05-11 by zaya-mc