ARM-software/psa-arch-tests #197: Test 50 does not call psa_get_key_attributes() API

zaya-mc, 2020-05-08:
In test 51, after checkpoint 14, tests checks attributes of the opened key but psa_get_key_attributes() not called before, so I would assume it needs to be called before checkpoint 14 first. 
```

+            /* Getting the attributes of the closed key should return error */
+            status = val->crypto_function(VAL_CRYPTO_GET_KEY_ATTRIBUTES, check1[i].key_handle,
+                &attributes);
+            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(14));
+
             val->crypto_function(VAL_CRYPTO_GET_KEY_TYPE, &attributes, &get_key_type);
-            TEST_ASSERT_EQUAL(get_key_type, check1[i].key_type, TEST_CHECKPOINT_NUM(14));
+            TEST_ASSERT_EQUAL(get_key_type, check1[i].key_type, TEST_CHECKPOINT_NUM(15));
 
             val->crypto_function(VAL_CRYPTO_GET_KEY_ID, &attributes, &get_key_id);
-            TEST_ASSERT_EQUAL(get_key_id, check1[i].key_id, TEST_CHECKPOINT_NUM(15));
+            TEST_ASSERT_EQUAL(get_key_id, check1[i].key_id, TEST_CHECKPOINT_NUM(16));
 
```

kotegowder, 2020-05-12:
Hi Murat,

Can you please check with the latest master once?
https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/dev_apis/crypto/test_c051/test_c051.c

This is taken care already. 
Please confirm at your end.

zaya-mc, 2020-05-13:
@kotegowder Yes, it seems it is fixed now. Happy to close.  Thank you.

closed 2020-05-13 by zaya-mc