ARM-software/psa-arch-tests #198: Different Expected Statuses for Seed/Label for HKDF

zaya-mc, 2020-05-13:
In Test 9, with PSA_ALG_HKDF(PSA_ALG_SHA_256), test expects PSA_ERROR_INVALID_ARGUMENT for Seed and Label steps when psa_key_derivation_input_bytes() called, 
which makes sense according to 
https://armmbed.github.io/mbed-crypto/html/api/ops/kdf.html#PSA_ALG_HKDF

The document says that only Key, Info and Salt available for HDKF.

But in test 20, with PSA_ALG_HKDF(PSA_ALG_SHA_256), test expects success for Seed and Label when psa_key_derivation_input_bytes() called.

```
            status = val->crypto_function(VAL_CRYPTO_KEY_DERIVATION_INPUT_BYTES, &operation,
                     check1[i].step, check1[i].data, check1[i].data_length);
            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(7));
```

Test 20 does not seem consistent with the API Spec?

zaya-mc, 2020-05-13:
The same issue for test 21 too.

kotegowder, 2020-05-25:
@zaya-mc : Agree with you. We will create pull request at our end.

kotegowder, 2020-06-01:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/209

closed 2020-06-01 by kotegowder