ARM-software/psa-arch-tests #199: Re-run of storage test results in failure - ITS

jainvikas8, 2020-05-13:
One of the tests has an effect of failing when its re-run - https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/psa_storage_testlist.md

This seems to be a bug, where the previous test didn't clear something or remanent of previous still remains in ITS making this test fail.

Please refer log - [Storage-tests.txt](https://github.com/ARM-software/psa-arch-tests/files/4622926/Storage-tests.txt)

An investigation is required on the PSA API ITS test suite.

PSA config
`    cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=GNUARM -DTARGET=tgt_dev_apis_tfm_musca_a -DSUITE=INTERNAL_TRUSTED_STORAGE \
            -DCPU_ARCH=armv8m_ml -DPSA_INCLUDE_PATHS=<include paths>`

kotegowder, 2020-05-14:
Can you take the previous working TF-M branch, apply recent check-in done to enable combined storage build support and the latest psa-arch-tests master?

Let us know if that works.

jainvikas8, 2020-05-14:
I have checked on both (before and after refactor of storage suite)

jainvikas8, 2020-05-14:
Tested on https://github.com/ARM-software/psa-arch-tests/commit/523fdb6795a9e45b07233e01972c8081322991a2
[storage -ITS - before psa suite change.txt](https://github.com/ARM-software/psa-arch-tests/files/4627771/storage.-ITS.-.before.psa.suite.change.txt)

kotegowder, 2020-05-14:
It is expected behavior. test_s002 trying to create a storage element with PSA_STORAGE_FLAG_WRITE_ONCE. If you re-run test, PSA_ITS_SET fails for the same UID.

jainvikas8, 2020-05-14:
The test needs to clear it before executing its intended purpose. It's not ideal that the user forcefully needs to clear the area of flash which was written once.

In the above test setup for the first run - I had to forcefully clear the area by
`srec_cat bl2/ext/mcuboot/mcuboot.bin -Binary -offset 0x200000 tfm_sign.bin -Binary -offset 0x220000 -fill 0xFF 0x420000 0x425000 -o tfm.hex -Intel`

If I don't use - `-fill 0xFF 0x420000 0x425000` then this test would fail everytime.

kotegowder, 2020-05-14:
Please refer to the behavior of PSA_STORAGE_FLAG_WRITE_ONCE in the specification at section 5.1.5 (page 15) https://github.com/ARM-software/psa-arch-tests/blob/master/api-specs/storage/v1.0/doc/IHI0087-PSA_Storage_API-1.0.0.pdf.

The data written with PSA_STORAGE_FLAG_WRITE_ONCE flag, cannot be modified or deleted.

Patater, 2020-05-14:
Is there a "reset device to factory default" PSA API we could run at the start of the arch tests to set the device into a good state?

MarcusJGStreets, 2020-05-14:
there is always a question as to what do you mean by default state. I beleive that the original thought was that write-once flag would be used for things like device id which need to be provisioned but which MUST not change. 

Obviously that makes testing problematic. 

I think that we can get round this by docmenting that the provisioning state can wipe the entire device - and permit return to provisioning for test devices - but notfor production devices. 

But I might need to think about that

Patater, 2020-05-14:
Which device lifecycle state would it be possible to write twice to write once values? "manufacturing" state?

athoelke, 2020-05-18:
The intention of `PSA_STORAGE_FLAG_WRITE_ONCE` is that it is not possible circumvent the access control policy that this requests **via the storage API**. Otherwise there would be no value in this attribute.

If the storage associated with an application, or the system can be reset in some way, for example, via an implementation-defined API - that is outside the scope of the PSA Storage API specification. However, a testing environment does benefit from that kind of API, it is just hard to determine what a generic, but secure version of that API would look like. This API must not be available to an attacker in a production system, so it requires some access control/authentication.

> In which device lifecycle state(s) would it be possible to write twice to write once values?

That makes testing the attribute meaningless, as it makes the attribute have no effect in the chosen lifecycle (which should probably be `PSA_LIFECYCLE_ASSEMBLY_AND_TEST`), so you cannot test that it works.

Possibly there might be value in an API or other mechanism that resets either the system, or perhaps just the current caller's PSA RoT state (storage and crypto perhaps); and which only works in `PSA_LIFECYCLE_ASSEMBLY_AND_TEST` lifecycle state.

Then the test suite checks the lifecycle state:
* If in `PSA_LIFECYCLE_ASSEMBLY_AND_TEST` it can reset the client's persistent state before running the suite.
* If not, then the affected tests need to deal gracefully with WRITE_ONCE test data that couldn't be removed. i.e. the test could report a "n/a" rather than a pass/fail.

Patater, 2020-05-19:
I am not sure that `PSA_LIFECYCLE_ASSEMBLY_AND_TEST` is meant for testing of the PSA API, rather for testing of the system you are building. Was this state intended to be the only one in which PSA API tests would run? If so, then how do you test that the PSA Lifecycle APIs are working in an integration-level test?

Should it be posible to test the PSA API for compliance using only PSA APIs? We don't have a PSA API yet that could "reset the client's persistent state before running the suite", so the test suite will have some additional, possibly non-PSA API requirements, in order to be runnable on your target. For example, a sort of HAL API along the lines you described that would erase all storage if the PSA lifecycle API. This HAL API would become a defacto part of PSA as any PSA implementation would want to be testable and thus would want to implement this. What value is provided to PSA implementers or users by not having a reset API?

kotegowder, 2020-05-22:
Having no architecturally defined reset API is out of scope for psa-arch-tests.
I can leave it to @athoelke and @MarcusJGStreets to comment on this.

However, there is way to skip the test by examining the presence of storage element which was written with PSA_STORAGE_FLAG_WRITE_ONCE.

Let us know, your feedback on that.

athoelke, 2020-05-22:
Having thought about this some more, I think that the test case should change to deal with the case that this item has already been written with the `PSA_STORAGE_FLAG_WRITE_ONCE`.

Instead of skipping the test, the test can still verify that a write-once item cannot be deleted or overwritten. The flow could be something like:

1. Get item info. If this fails with DOES_NOT_EXIST goto step 4, otherwise check it succeeds.
2. If item flags include WRITE_ONCE, goto step 6
3. Delete item, check this succeeds.
4. Set item with WRITE_ONCE flag, check this is successful.
5. Get item info. Check this is successful.
6. Check the item info flags are correct (WRITE_ONCE is set).
7. Get the item data and check the content is correct.
8. Try to set the item. Check this fails with `PSA_ERROR_NOT_PERMITTED`.
9. Try to delete the item. Check this fails with `PSA_ERROR_NOT_PERMITTED`.

Step 2 deals with a pre-existing WRITE-ONCE item. It tests that the item values were written by this test suite previously - if that is a problem, it could jump to step 8 instead.

MarcusJGStreets, 2020-05-22:
You do not have a step to set it without WRITE_ONCE and then delete it.
Though that could be a different test of course.

Also it does not catch the case where a previous version of code could write an item with write once, but some glitch in this version means either now writing fails of the flag is not set.

But you can only test that if you are prepared to either exhaust the memory – or have a way of defeating the setting.

You could create a ITS that does not honour the bit interleave the two
So version 1 is good
Version 2 is bad and you use that to reset
Then version 3 is the next candidate good version
Then sign bad as version 4
Etc.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

kotegowder, 2020-05-28:
I'm fine with extended test logic as mentioned by Andrew for the storage element which was created using PSA_STORAGE_FLAG_WRITE_ONCE, instead of just skipping.

@Patater : Let us know your feedback.

Patater, 2020-05-28:
@kotegowder I'm fine with this testing approach, since it lets us run the tests repeatedly (useful when doing the initial board bringup with TF-M and PSA)

kotegowder, 2020-06-01:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/209

closed 2020-06-01 by kotegowder