ARM-software/psa-arch-tests #205: Usability of arch tests with different target names.

jainvikas8, 2020-05-22:
Is there a reason why we have 2 different target name?
For example: `ARM_MUSCA_A1` has  **tgt_dev_apis_tfm_musca_a**, **tgt_ff_tfm_musca_a** 
Suites: `CRYPTO, INITIAL_ATTESTATION/INTERNAL_TRUSTED_STORAGE/PROTECTED_STORAGE/STORAGE` is used for **tgt_dev_apis_tfm_musca_a**.
In the build system we use `IPC` as a suite for **tgt_ff_tfm_musca_a**, so does it mean:
* There may be new suites for **tgt_ff_tfm_musca_a** in the near future?
* Are there any linker script changes for **tgt_ff_tfm_musca_a** compared to **tgt_dev_apis_tfm_musca_a**?
* Any other reason?

If none, a simple linking could be made in build system to link
`ARM_MUSCA_A1` with a particular suite if applicable and return an error if not?
For example:
`cmake ../ -G"Unix Makefiles" -DTOOLCHAIN=GNUARM -DTARGET=ARM_MUSCA_A1 -DSUITE=<dev_apis/ff>   -DCPU_ARCH=armv8m_ml -DPSA_INCLUDE_PATHS=<paths>`

This will improve the usability and avoid confusion regarding different target names present for a target

kotegowder, 2020-05-22:
There are different reasons for keeping them separate to ease the partner while creating their own target support for psa-arch-tests. 

1. The L1 Functional API certification involves running only Functional APIs and not Firmware Framework, which is still not part of to do list.
2. There are quite a number of Functional API users compare to FF. 
3. When partner go for L1 certification for their target, we want to ensure they only need to port what is necessary and not worry about IPC related target files and vice versa.
4. For Level 2 isolation, it is not mandatory that partner has to implement the PSA-M FF specified IPC APIs and we do not want them to see any IPC related files. 

Also, there is marketing input as well for this structuring as it focuses more on Functional APIs at present.

kotegowder, 2020-05-28:
psa-arch-tests encourages partner to maintain the target that works best in their convenience.

closed 2020-05-28 by kotegowder