ARM-software/psa-arch-tests #206: PS 11 test expected error

zaya-mc, 2020-05-26:
In the Protected Storage 11 tests, VAL_TEST_IDX2 step creates a record using psa_ps_set() function, and in the next step(VAL_TEST_IDX3, "[Check 2] Call create API with length different than original\n"), tests calls psa_ps_create() function with the same UID and expects PSA_ERROR_INVALID_ARGUMENT. 

In the psa_ps_create() function header, it says if the UID exists, it should return PSA_ERROR_ALREADY_EXISTS

```
 * \retval PSA_ERROR_ALREADY_EXISTS        Storage for the specified uid
 *                                         already exists
```

So, which error code is correct?

zaya-mc, 2020-05-26:
In the PSA Spec, it says it needs to return ALREADY_EXIST error

```
This function cannot be used to replace an existing asset, and attempting to do so will return PSA_ERROR_ALREADY_EXISTS.
```

kotegowder, 2020-05-27:
@zaya-mc : The current logic in test_s011, trying to call psa_ps_create( ) using PSA_STORAGE_FLAG_WRITE_ONCE where psa_ps_set( ) is called with PSA_STORAGE_FLAG_NONE in the previous step.

The error code PSA_ERROR_INVALID_ARGUMENT is because of the different 'create_flags' parameter between psa_ps_set( ) and psa_ps_create( ) calls.

However, I see there is statement 
'If the PSA_STORAGE_FLAG_WRITE_ONCE flag is passed, psa_ps_create will return PSA_ERROR_NOT_SUPPORTED.' . I believe the check should have been against PSA_ERROR_NOT_SUPPORTED instead of PSA_ERROR_INVALID_ARGUMENT.

Some of the tests which involve these optional APIs (psa_ps_create, psa_ps_set_extended) are currently not exercised due to the missing implementation from our reference platform.

@jf549 : Can we fix the test with this assumption of PSA_ERROR_NOT_SUPPORTED ?

zaya-mc, 2020-05-27:
Hi @kotegowder, yes there are three conditions the API needs to consider, there must be order for error checking. I believe ALREADY_EXIST error should higher priority then PSA_ERROR_INVALID_ARGUMENT error of PSA_STORAGE_FLAG_XXX check, because if the UID is there, we should not accept any check (e.g. Flag) on it.

jf549, 2020-05-27:
psa_ps_create() should return PSA_ERROR_ALREADY_EXISTS if it is called on a uid that already exists and should return PSA_ERROR_NOT_SUPPORTED if it is called with the PSA_STORAGE_WRITE_ONCE_FLAG create_flag.

If it is called with the PSA_STORAGE_WRITE_ONCE_FLAG on an existing uid, then either of the two errors would be valid and could be returned. I don't believe that the spec defines any order of precedence for error codes for cases where more than one occurs.

kotegowder, 2020-05-27:
Going ahead with PSA_ERROR_NOT_SUPPORTED for now, as the specification explicitly mention the condition with PSA_STORAGE_WRITE_ONCE_FLAG.

athoelke, 2020-05-28:
> If it is called with the PSA_STORAGE_WRITE_ONCE_FLAG on an existing uid, then either of the two errors would be valid and could be returned. I don't believe that the spec defines any order of precedence for error codes for cases where more than one occurs.

The precedence of errors is implementation-defined in a case where a call is not valid for more than one reason. This is more clearly described in the PSA Crypto API specification, but is also the approach that should be taken with testing the Attestation and Storage APIs.

The reason for the flexibility is that this allows constrained implementations to optimize the validation checks in a way that matches the underlying storage design. In practice, both of these errors are programming errors in the application, and should not be expected to occur in correctly written code - thus there is limited value in specifying the precedence of the errors.

kotegowder, 2020-06-01:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/209

closed 2020-06-01 by kotegowder

zaya-mc, 2020-06-01:
@kotegowder thank you for the fix but there was one more remaining. 

In the test case 10

```
{
    /* Again call create API with different length */
    VAL_TEST_IDX10, {VAL_API_UNUSED, VAL_PS_CREATE}, PSA_ERROR_INVALID_ARGUMENT
},
```

it expects PSA_ERROR_INVALID_ARGUMENT from psa_ps_create() function for an existing UID. 
I would expect PSA_ERROR_ALREADY_EXISTS error.

zaya-mc, 2020-06-01:
Also, what about the PS test 13; 

```
 {
     /* Call Create API for existing UID with same parameters */
    VAL_TEST_IDX16, {VAL_API_UNUSED, VAL_PS_CREATE}, PSA_SUCCESS
 },
```

I would expect PSA_ERROR_ALREADY_EXISTS too here.

kotegowder, 2020-06-01:
Will update the master with the changes.