ARM-software/psa-arch-tests #208: Warnings

zaya-mc, 2020-05-27:
We are running the PSA Tests in different environments such as Visual Studio, and it may be minor but there are warnings you might want to fix. 

You can see the file, line and the column info below, they may change but we are working with the following commit
 SHA : 0549c27ef77809cdd12ad84c26e9dd5af696f8a8

```
\psa-arch-tests\api-tests\dev_apis\crypto\test_c017\test_c017.c(50,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c020\test_c020.c(51,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c021\test_c021.c(55,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c024\test_c024.c(35,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c025\test_c025.c(35,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c027\test_c027.c(132,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c034\test_c034.c(53,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c036\test_c036.c(147,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c039\test_c039.c(34,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c040\test_c040.c(34,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c044\test_c044.c(59,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c054\test_c054.c(97,23): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\crypto\test_c060\test_c060.c(102,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\initial_attestation\test_a001\test_a001.c(41,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\initial_attestation\test_a001\test_a001.c(93,19): warning C4018: '<': signed/unsigned mismatch
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(43,89): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(49,94): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(54,92): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(88,89): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(94,94): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(99,92): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(112,61): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(118,84): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(124,89): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s001\test_s001.c(129,87): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(62,93): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(81,94): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(105,87): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(111,94): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(129,92): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s002\test_s002.c(153,99): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s003\test_s003.c(53,51): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s003\test_s003.c(59,92): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s003\test_s003.c(66,44): warning C4244: '=': conversion from 'psa_storage_uid_t' to 'uint32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s003\test_s003.c(67,49): warning C4244: '=': conversion from 'psa_storage_uid_t' to 'uint32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s003\test_s003.c(74,60): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s011\test_s011.c(43,98): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
\psa-arch-tests\api-tests\dev_apis\storage\test_s011\test_s011.c(103,98): warning C4244: 'function': conversion from 'psa_storage_uid_t' to 'int32_t', possible loss of data
```

kotegowder, 2020-06-14:
@zaya-mc : Can you please check if PR https://github.com/ARM-software/psa-arch-tests/pull/221 addresses this issue.

zaya-mc, 2020-06-15:
Hi @kotegowder, I was getting these warnings with Visual Studio build, and I tested with VS and it seems fine. I don't have a chance to test with the real environment. 
Just I am wondering about __UNUSED flag for unused parameters, is that portable? 
We might use compiler switches to creates portable UNUSED macros or we might explicitly call like below

(void)unusedParam;

kotegowder, 2020-06-15:
The changes are tested against the compilers that psa-arch-tests support today(arm-none-eabi-gcc, armclang and host gcc on linux).

Do you see any problem in terms of supporting through macro for any other compiler support that may come in future ?

zaya-mc, 2020-06-15:
What about ARMCC (Arm MDK)? For now, it should be fine for us but you might consider creating flexible/ portable solution. 
We are happy to close the ticket. Thanks.

kotegowder, 2020-06-17:
For now closing the ticket.
Will be re-looked into, when there is going to be more compiler support in future.

closed 2020-06-17 by kotegowder