ARM-software/psa-arch-tests #210: Different Expected Statuses for psa_aead_set_lengths

zaya-mc, 2020-05-27:
In test crypto 56 and in checkpoint 6, tests expect PSA_ERROR_INVALID_ARGUMENT for "Zero ad size" case. 

```
        status = val->crypto_function(VAL_CRYPTO_AEAD_SET_LENGTHS, &operation, check1[i].ad_length,
                 check1[i].plaintext_length);
        TEST_ASSERT_EQUAL(status, check1[i].expected_status, TEST_CHECKPOINT_NUM(6));
```

But in crypto 57, it expects explicitly success for the "Zero ad size" case. 

```
        status = val->crypto_function(VAL_CRYPTO_AEAD_SET_LENGTHS, &operation, check1[i].ad_length,
                 check1[i].plaintext_length);
        TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(6));
```

zaya-mc, 2020-05-27:
Can the AD size be Zero? I don't see any information about it in the spec. In our system, we accept zero length AD (or any size).

zaya-mc, 2020-05-27:
The same issue with the zero plain text case

zaya-mc, 2020-05-27:
For test case 57, we need to be careful while fixing this issue. If psa_aead_set_lengths() fails it needs to abort the operation and in the next checkpoint, it will call the psa_aead_update_ad()  and it will check with the expected error state. In case of abort, the error will be always PSA_ERROR_BAD_STATE.

zaya-mc, 2020-05-28:
The same issue in Test 58 for "Zero ad size" case.

kotegowder, 2020-06-02:
I just checked internally, psa_eaed_set_lengths with ad_length=plaintext_length=0 should result is PSA_SUCCESS.

Also about ad_length=plaintext_length=1 for psa_aead_set_lengths and 
input_length=1 for psa_aead_update_ad should be a valid case as well.

We will re-work on the test logic.

kotegowder, 2020-06-09:
We have an open issue for the unimplemented multi-part AEAD APIs.
https://github.com/ARMmbed/mbedtls/issues/3257

The changes for this issue comes in our next release, which we could verify on TF-M(mbedtls) reference. 

On timeline front, it is TBD at present.

kotegowder, 2020-08-24:
@zaya-mc : We would like to know is it fine if the fix be made available as part of our next release(probably in Nov) or in any way these are blocking for your any activities. Let us know!

zaya-mc, 2020-08-25:
@kotegowder thank you for your kind interest. Actually, we have disabled some cases and we also have minor fixes to pass tests. It would be nice if you fix them but we can wait until November. Let us know when they are fixed. Thank you
Murat

jk-arm, 2021-07-16:
@zaya-mc, we had recently released v1.2(tag : v21.07_API1.2_ADAC_ALPHA) and the above-mentioned test changes are addressed, unfortunately, our reference platform still missing the AEAD feature implementation, so these changes are not validated.

closed 2021-07-21 by jk-arm