ARM-software/psa-arch-tests #211: 64 Bit Variables in Variadic Functions

zaya-mc, 2020-05-28:
We are experiencing some problem with 64 bits variables in variadic functions. 
The same code can behave differently in different environments.

Example please see the below code (Protected Storage Test 12); 

```
    val->print(PRINT_TEST, "[Check 9] Call remove API with UID 0\n", 0);
    status = STORAGE_FUNCTION(s012_data[VAL_TEST_IDX13].api[fCode], 0);
    TEST_ASSERT_EQUAL(status, s012_data[VAL_TEST_IDX13].status, TEST_CHECKPOINT_NUM(14));
```

In the above function we pass UID as Zero. It is a variadic function and in a 32bit system, the compiler should promote "Zero" to 32bit. 
But once we get it from the va_list, we get is as psa_storage_uid_t variable which is 64 bit.

         `uid = va_arg(valist, psa_storage_uid_t);`

I don't see a problem on 64bit Windows machine (probably promotes to 64 bit integer), but when I run the same code on 32 bit embedded environment, I got something random. 

I would recommend explicit cast 
 - `(psa_storage_uid_t )0`
or using a literal after the constant values to cast the values to 64 bit. 
- `0LL`

kotegowder, 2020-05-28:
Thank you for bringing this up.
We shall create PR for this.

kotegowder, 2020-06-01:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/209

closed 2020-06-01 by kotegowder