ARM-software/psa-arch-tests #214: Expected Error for set_extended function in PS test 16

zaya-mc, 2020-06-01:
In the PSA test 16, the test case 10, 
the test tries to call psa_ps_set_extended with the size higher than capacity and expects PSA_ERROR_STORAGE_FAILURE. 

```
 {
     /* Set_extended on old capacity should not work */
    VAL_TEST_IDX10, {VAL_API_UNUSED, VAL_PS_SET_EXTENDED}, PSA_ERROR_STORAGE_FAILURE
 },
```

As I understand PSA_ERROR_STORAGE_FAILURE, is about low level IO error. 

```
 * \retval PSA_ERROR_STORAGE_FAILURE    The data was not written correctly in
 *                                      the physical storage

```

But herein I would expect PSA_ERROR_INVALID_ARGUMENT error while the argument is invalid.

athoelke, 2020-06-01:
The expected return code should be `PSA_ERROR_INVALID_ARGUMENT`, though the documentation is not entirely clear as that status code describes pre-conditions 'above' in the documentation. The pre-conditions are actually below this:

>This function can overwrite existing data and/or extend it up to the capacity for the `uid` specified in `psa_ps_create()`, but cannot create gaps. That is, it has preconditions:
>
>* `data_offset <= size`
>* `data_offset + data_length <= capacity`

kotegowder, 2020-06-12:
Fixed with https://github.com/ARM-software/psa-arch-tests/pull/219

closed 2020-06-12 by kotegowder