ARM-software/psa-arch-tests #215: Missing IV data in for psa_cipher_encrypt/psa_cipher_decrypt functions in test 48/49

zaya-mc, 2020-06-01:
In the psa_cipher_encrypt() function, it says the output should include IV. 

```
 * \param[out] output           Buffer where the output is to be written.
 *                              The output contains the IV followed by
 *                              the ciphertext proper.
```

It is fine, we can create an IV and return it.

But for the psa_cipher_encrypt() tests it says the input includes IV

```
 * \param[in] input             Buffer containing the message to decrypt.
 *                              This consists of the IV followed by the
 *                              ciphertext proper.
```

But, when we see the test data in test 49, the input length is 16. 
So, does it mean 16 Bytes IV and 0-byte input, or the IV is not there?

We might also align test 48 too because if we need to include IV in out buffer, we might need to revisit the expected output length too.

zaya-mc, 2020-06-01:
The side effect may occur when we fix this issue. 
For example; "Test psa_cipher_decrypt - Decrypt - AES CBC_NO_PADDING\n" case needs to be revisited too. 

```
{"Test psa_cipher_decrypt - Decrypt - AES CBC_NO_PADDING\n", PSA_KEY_TYPE_AES,
{0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09,
 0xcf, 0x4f, 0x3c}, AES_16B_KEY_SIZE,
 PSA_KEY_USAGE_DECRYPT, PSA_ALG_CBC_NO_PADDING,
{0xA0, 0x76, 0xEC, 0x9D, 0xFB, 0xE4, 0x7D, 0x52, 0xAF, 0xC3, 0x57, 0x33, 0x6F,
 0x20, 0x74, 0x3B},
 16, 10,
{0x6b, 0xc1, 0xbe, 0xe2, 0x2e, 0x40, 0x9f, 0x96, 0xe9, 0x3d, 0x7e, 0x11, 0x73,
 0x93, 0x17, 0x2a}, 0, PSA_ERROR_BUFFER_TOO_SMALL
},
```

At the moment, in our custom implementation, we extract IV from the input. In this case, input length is 16 while output size is 10. So, once we extract IV from input buffer, we have zero-byte cipher text and output for zero byte cipher will be zero which is less then 10 byte output size, so it will pass while we expect PSA_ERROR_BUFFER_TOO_SMALL. 
So, herein input byte should be updated to 32 byte.

zaya-mc, 2020-06-01:
The other case which needs to be updated is that ""Test psa_cipher_decrypt - Decrypt - AES CTR (short input)\n""

```
{"Test psa_cipher_decrypt - Decrypt - AES CTR (short input)\n", PSA_KEY_TYPE_AES,
{0x2b, 0x7e, 0x15, 0x16, 0x28, 0xae, 0xd2, 0xa6, 0xab, 0xf7, 0x15, 0x88, 0x09,
 0xcf, 0x4f, 0x3c}, AES_16B_KEY_SIZE,
 PSA_KEY_USAGE_DECRYPT, PSA_ALG_CTR,
{0x8f, 0x94, 0x08, 0xfe, 0x80, 0xa8, 0x1d, 0x3e, 0x81, 0x3d, 0xa3, 0xc7, 0xb0,
 0xb2, 0xbd}, 15, SIZE_32B,
{0x6b, 0xc1, 0xbe, 0xe2, 0x2e, 0x40, 0x9f, 0x96, 0xe9, 0x3d, 0x7e, 0x11, 0x73,
 0x93, 0x17}, 0, PSA_SUCCESS
},
```

Herein once we try to extract IV from the input, the input len (15 bytes) is less then IV size so we return error but this function expects success so the input buffer needs to include IV here

jk-arm, 2021-07-28:
this triggered an internal discussion and is yet to get a resolution, we will update you back once we have a concrete answer.

athoelke, 2021-08-09:
Apologies for some delay in looking at this, I have been on vacation since I was notified about this question.

### Specification

The specification describes the output parameter of `psa_cipher_decrypt()` as "Buffer where the plaintext is to be written". Perhaps it could be clearer by repeating in the description that the output is just the plaintext, following the description of the input buffer? - but is the current spec ambiguous?

Early versions of the specification (beta-1 and beta-2) were undecided about the handling of IV in the multipart and one-shot cipher APIs, but the current approach has been present since 1.0-beta-3.

### Mbedtls

Mbedtls has only supported the single-part/one-shot cipher functions since v2.27.0/v3.0 which were released on Jul 7 this year. So I suspect that tests 48 and 49 have not been fully exercised until recently.

The mbedtls implementation (see [`cipher_decrypt()` in psa_crypto_cipher.c](https://github.com/ARMmbed/mbedtls/blob/8df2f8e7b9c7bb9390ac74bb7bace27edca81a2b/library/psa_crypto_cipher.c#L503)) appears to follow the specification intention correctly: extracting the default-length IV from the input, and not including the IV in the decryption into the output buffer.

### Test suite

Given the lack of reference implementation for testing this functionality - it seems that a review of the test suite data for 48 and 49 should be done against the specification and the mbedtls implementation.

athoelke, 2021-08-09:
Looking at the test suite 49 data: some of the tests have an `expected_output_length` of zero, but a non-zero `output_size` and a `expected_output` buffer containing a sequence of bytes that does not always match `output_size` or `expected_output_length`.

In `Test psa_cipher_decrypt - Decrypt - AES CBC_NO_PADDING`, for example, the `output_size` of 10 and `expected_output` buffer have no real effect on the test result because the `expected_output_length` is zero, which is correct for a 16 byte input which is entirely consumed as the IV.

athoelke, 2021-08-09:
If you call `psa_cipher_encrypt()` with that plain text, you cannot specify the IV. In this case the implementation must generate a random 16-byte IV, use it to encrypt the plain text, and then output 32 bytes: the 16-byte IV followed by the 16-byte ciphertext.

If you want to use a specific IV for AES-CBC encryption, then you need to use the cipher operation, so that you can call `psa_cipher_set_iv()` with the chosen IV. In this case, the output of the `psa_cipher_update()` and `psa_cipher_finish()` calls will not include the IV, and only be the 16-byte ciphertext (the last 16 bytes of your example).

With your example, you could then pass the key and the 32-byte IV+ciphertext string to `psa_cipher_decrypt()` and it should return the 16-byte plaintext value.

jk-arm, 2021-08-09:
Thanks Andrew,This clarifies,

from testcase data point of it,  we cant have the expected ciphered data in a single step encrypt because the IV is calculated internally.
Let me make the changes accordingly.

Regards,
JK

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

jk-arm, 2021-08-09:
fixes can be seen in branch : https://github.com/ARM-software/psa-arch-tests/tree/key_derivation

athoelke, 2021-08-10:
Hi @jk-arm, the changes in 3426ad3 are not the ideal way to test the one-shot encryption:

* The wrong-usage-flag test "Test psa_cipher_encrypt - Decrypt - AES CBC_NO_PADDING" should not need the addition of the operation object and calling `psa_cipher_encrypt_setup()` to check for `PSA_ERROR_NOT_PERMITTED` prior to calling `psa_cipher_encrypt()`. There seems to be no other reason to add the operation object in the code at present/ (but see my suggestions below)
* Removing the comparison of the ciphertext removes the ability for the test suite to verify that the cipher operation is performing a correct encryption.

I think that we should continue to check that the cipher text output from `psa_cipher_encrypt()` is correct, but because of the randomized IV, this will require the use of either the multi-part cipher operation encryption or decryption code. I assume that the test suite already tests multi-part cipher encryption with known IV+plaintext+ciphertest test vectors - and thus we can established confidence that the multi-part operation works as specified.

This is my suggestion for testing the one-shot cipher API:

Testing `psa_cipher_decrypt()` can be done with known test vectors using IV+ciphertext as input and plaintext as output.

Testing `psa_cipher_encrypt()` can be done by comparing the output of the single-part function, with the same encryption done by the multi-part operation:

   1. The test vectors include: key attributes+data, algorithm, input plaintext data+length, output_size, expected output_length, expected return status. But no expected output value. This allows you to check most of the negative cases of invalid inputs.
   2. Import the key as already done.
   3. Use the key and input data in a call to `psa_cipher_encrypt()`, storing the output in `output_1[]` (or some better named buffer). Check that the return status and the output length match the expected values.
   4. If the result of the one-shot encryption was not `PSA_SUCCESS`, then the test is finished, and go to step 11 to clean up.
   5. Initialize and setup a `psa_cipher_operation_t` for encryption using the same key and algorithm. This should return `PSA_SUCCESS`.
   6. Use `psa_cipher_set_iv()` to set the encryption IV using the first `PSA_CIPHER_IV_LENGTH(key_type, algorithm)` bytes of `output_1[]`. This should return `PSA_SUCCESS`.
   7. Use `psa_cipher_update()` to pass the plaintext (the input in the test vector used in step 3), storing the output in `output_2[]`. This should return `PSA_SUCCESS`.
   8. Use `psa_cipher_finish()` to complete the encryption, appending the output to `output_2[]` (use the output_length from step 7). This should return `PSA_SUCCESS`.
   9. Call `psa_cipher_abort()` to clean up the operation.
   10. Compare the ciphertext in `output_1[]` (starting at offset `PSA_CIPHER_IV_LENGTH(key_type, algorithm)`) with the ciphertext in `output_2[]` (combined result of step 7 & 8) - these should be an exact match.
   11. Destroy the key

jk-arm, 2021-08-10:
HI Andrew,
Thanks for the detailed suggestion, we will try out these and come back.

Regards,
JK

jk-arm, 2021-08-11:
@athoelke , this testcase is not intend to test the multipart encryption, 
1. wrong usage flag suggestion will be taken care.
2. the use of psa_cipher_encrypt_setup might have led to confusion, for verifying the output ciphertext in one-shot cipher, I have some comments 
       i.   As you have suggested let's collect the one-shot cipher output
      ii.  then take the IV from the output buffer and perform the multipart encryption
     iii.  compare the both output
    **Question:**
       in the above suggestion, i am not seeing any predefined test data to work with. we are trying to generate the data on the fly. this might lead us to other problems like not able to identify the problem on encryption( lets say the encryption code has the bug and we are generating testdata through one-shot and verifying the same with multi-part, here the wrongly encoded data gets tested and the verdict will be success.)

hope i made my point clear. **FYI,** we have other test cases to check the multipart.

jk-arm, 2021-08-11:
> I understand that using multi-part to validate one-shot could lead to not detecting an encryption error. However, we can run separate test on multi-part which provide a full input+output test vector. So if that test suite provides confidence in the correctness of the cipher encryption multi-part operation, then we can use this to validate the one-shot functionality.
>
> An alternative is to use one-shot decryption to validate one-shot encryption - but this provides less scope for checking that the encryption output really is made up of the IV + ciphertext as described in the spec. In this case, we expect to validate the correctness of one-shot decryption via a full set of input+output test vectors (there is no randomness involved in decryption).

HI Andrew,

I get your intention, we should definitely make an alternative testcase for your suggestion. I will make this suggestion into our scenario document and try covering in the upcoming release.
But for now this testcase can go as it is.

Hope you are with me in this.

Regards,
JK

athoelke, 2021-08-11:
The current test case makes no attempt to check that the encryption is done correctly - only that the inputs are handled as described and the length of the output matches that expected for the key, algorithm and input data.

So it is better than nothing, but it is definitely incomplete if it does not check that the ciphertext is correct.

jk-arm, 2021-08-11:
HI Andrew,
thank you for your input, l agree, let me try incorporate the suggestion in the testcase 48 itself.

Regards,
JK

bensze01, 2021-11-09:
This is still an issue with `v21.10_API1.3_ADAC_ALPHA-1` - The DES /3DES cases in test_c048 and test_c049 are not handled properly (The IV vectors need to be 8 bytes long in those cases).

Also, the code in those tests still doesn't attempt to verify that the returned values are correct, it only checks if the length is the same.

jk-arm, 2021-11-11:
> This is still an issue with `v21.10_API1.3_ADAC_ALPHA-1` - The DES /3DES cases in test_c048 and test_c049 are not handled properly (The IV vectors need to be 8 bytes long in those cases).
>
> Also, the [...]

our reference platform does not support the DES ( less secure) we could not able to verify, please feel free to submit a pull request, we can review the same and approve.