ARM-software/psa-arch-tests #217: Expected Signature for Asymmetric Sign

zaya-mc, 2020-06-03:
Hi all, 

For Asymmetric(RSA/ECC) sign tests, there are expected signatures compared with actuals but algorithms use random so for each signature, we should get different signatures so I would not expect a signature comparison. 

For example in test 41
```
/* Check if the output matches with the expected data */
TEST_ASSERT_MEMCMP(signature, check1[i].expected_signature, length, TEST_CHECKPOINT_NUM(7));
```

kotegowder, 2020-07-22:
@zaya-mc : Please expect the updates for these open issues to come in our next release.

jk-arm, 2021-07-19:
@zaya-mc, we are doing the internal discussion on this to make it more relavent, we will come back shortly.

jk-arm, 2021-08-06:
we should be changing the memcmp to verify api.

jk-arm, 2021-08-10:
this is fixed in internal Gerrit, next release will have this changes on GitHub