ARM-software/psa-arch-tests #218: ECC Key Pair Test Data includes only Private Key

zaya-mc, 2020-06-04:
When PSA Tests import an ECC Key pair, it imports only <Key_Len> private key which is fine, we can reproduce a public key `<2 * Key_Len>` using the private key. 

Some solutions may explicitly need public key during the ECC signature verify. So, could we provide 
"`3 * <Key_Length>`" ECC key(Private+Public) if the Key Type is "`PSA_KEY_TYPE_ECC_KEY_PAIR(...)`".

zaya-mc, 2020-06-22:
@kotegowder have you had a chance to think about the issue mentioned above. Thanks.

kotegowder, 2020-06-23:
The key pair interpretation vary based on the algorithm.
For some it is just private part but still qualifies to be a key pair.

@athoelke : Can you confirm this ?

athoelke, 2020-06-23:
The default format for importing keys with `psa_import_key()` is the format defined for key export in `psa_export_key()`. For ECC key-pairs, this contains just the private key component encoded appropriately for the curve family, because reconstruction of the public key (if required for verification, encryption or export as a public key) is simple.

`psa_import_key()` is permitted to accept **additional** implementation-specific formats, so long as the additional formats do not create ambiguity for interpreting the data.

Supporting both <private-key> and <private+public-key> formats for importing an ECC key pair meets this requirement as the length identifies the format. However, the implementation should consider verifying that a <private+public-key> is indeed a key pair before using it.

As adding a format like this is an implementation-specific option, testing this would be in the scope of implementation-specific testing.

jk-arm, 2021-07-19:
@zaya-mc , can i consider this for closure?

jk-arm, 2021-07-21:
since this is implementation-specific as Andrew mentioned, I am closing this ticket.

closed 2021-07-21 by jk-arm