ARM-software/psa-arch-tests #220: Support for registering PSA Secure Element drivers for PSA Crypto tests

hug-dev, 2020-06-08:
### Feature request

The [PSA Secure Element interface](https://github.com/ARMmbed/mbedtls/blob/development/include/psa/crypto_se_driver.h) allows to defer the key management and crypto operations execution to an external entity than the PSA Crypto implementation.
An application wanting to use a Secure Element driver instead of the main crypto implementation, has to call `psa_register_se_driver` (before `psa_init`) to register it with a specific lifetime value. Keys generated with the same lifetime will be handled by the SE (Secure Element) driver instead of the main implementation.

This issue is to add support for Secure Element drivers so that the PSA Crypto tests can also test the execution going through the SE driver.

I see three changes needed in the codebase to support them:
1. Support in the build system a SE driver option to add conditional compilation and link to the SE driver implementation structure.
2. Add a call to `psa_register_se_driver` before any call to `psa_init`. The driver must be registered only once.
2. Replace in all tests (or a selection?) the keys' lifetime with the same lifetime the driver was registered with so that keys are generated by the SE driver and crypto operations with those keys are routed to the SE driver.

### Context of my request

I am working on the [Parsec project](https://github.com/parallaxsecond/parsec) and we developped a [Parsec Secure Element driver](https://github.com/parallaxsecond/parsec-se-driver#how-to-build-and-use-the-driver) so that developpers can use in their C applications Mbed Crypto and register this Parsec SE driver so that it uses Parsec.
It would be highly beneficial for us to re-use the Functional API tests on PSA Crypto in this repository in order to test Parsec. That would be done with the `tgt_dev_apis_stdc` target.

I am going to try to investigate on how to implement this issue locally but if you have any idea on what would be the best way to do it, I am then happy to make a PR with my changes to upstream this feature.

kotegowder, 2020-06-15:
Had discuss with @hug-dev on the structure of psa-arch-tests.
Suggestion is to lean towards a separate target for PSA Arch tests + PARSEC work, instead of diluting the existing tgt_dev_apis_stdc.

hug-dev, 2020-06-15:
I did some more investigations and saw that the different changes needed were:

* In the `main.c` file:

```diff
diff --git a/api-tests/platform/targets/tgt_dev_apis_stdc/nspe/main.c b/api-tests/platform/targets/tgt_dev_apis_stdc/nspe/main.c
index 34c28e1..c999ee2 100644
--- a/api-tests/platform/targets/tgt_dev_apis_stdc/nspe/main.c
+++ b/api-tests/platform/targets/tgt_dev_apis_stdc/nspe/main.c
@@ -16,6 +16,9 @@
 **/

 #include <stdint.h>
+#include "psa/crypto_se_driver.h"
+#include "parsec_se_driver.h"

 int32_t val_entry(void);

@@ -27,5 +30,11 @@ int32_t val_entry(void);
 **/
 int main(int argc, char **argv)
 {
+    psa_status_t status = psa_register_se_driver((psa_key_lifetime_t)2,
+        &PARSEC_TPM_DIRECT_SE_DRIVER);
+    if (status != PSA_SUCCESS) {
+        return status;
+    }
     return val_entry();
 }
```
`2` is  the new lifetime introduced with this registration to be used with the SE driver. Every key assigned to this lifetime will be redirected to the Parsec SE driver.
* In the `target.cmake` file:
```diff
diff --git a/api-tests/platform/targets/tgt_dev_apis_stdc/target.cmake b/api-tests/platform/targets/tgt_dev_apis_stdc/target.cmake
index 7d002d1..920bd8e 100644
--- a/api-tests/platform/targets/tgt_dev_apis_stdc/target.cmake
+++ b/api-tests/platform/targets/tgt_dev_apis_stdc/target.cmake
@@ -51,7 +51,15 @@ function(_create_psa_stdc_exe _exe_name _api_dir)
        )

        add_executable(${EXE_NAME} ${EXE_SRC})
+       target_include_directories(${EXE_NAME} PRIVATE ${PARSEC_SE_DRIVER_INCLUDE_PATHS})
+       target_link_libraries(${EXE_NAME} -lpthread -ldl -lm ${PARSEC_SE_DRIVER_LIB_FILENAME})
+       target_include_directories(${EXE_NAME} PRIVATE ${PSA_INCLUDE_PATHS})
        target_link_libraries(${EXE_NAME} ${EXE_LIBS} ${PSA_CRYPTO_LIB_FILENAME} ${PSA_STORAGE_LIB_FILENAME})

@@ -132,3 +140,5 @@ target_include_directories(${PSA_TARGET_PAL_NSPE_LIB} PRIVATE
        ${PSA_ROOT_DIR}/platform/targets/common/nspe/internal_trusted_storage
        ${PSA_ROOT_DIR}/platform/targets/${TARGET}/nspe
 )
+
+target_include_directories(${PSA_TARGET_PAL_NSPE_LIB} PRIVATE ${PARSEC_SE_DRIVER_INCLUDE_PATHS})
```
With the new environment variables defined, that changes the compilation to:
```
cmake ../ -DTARGET=tgt_dev_apis_stdc -DTOOLCHAIN=HOST_GCC -DSUITE=CRYPTO -DPSA_INCLUDE_PATHS="/home/hugdev01/dev/psa-arch-tests/api-tests/mbedtls/include" -DPSA_CRYPTO_LIB_FILENAME=/home/hugdev01/dev/psa-arch-tests/api-tests/mbedtls/library/libmbedcrypto.a -DPARSEC_SE_DRIVER_INCLUDE_PATHS="/home/hugdev01/dev/parallaxsecond/parsec-se-driver/include" -DPARSEC_SE_DRIVER_LIB_FILENAME="/home/hugdev01/dev/parallaxsecond/parsec-se-driver/target/debug/libparsec_tpm_direct_se_driver.a"
```
* In every test that I want to execute on the SE driver, I have to modify the key's attribute with:
```
val->crypto_function(VAL_CRYPTO_SET_KEY_ID, &attributes, (psa_key_id_t)(i + 1));
val->crypto_function(VAL_CRYPTO_SET_KEY_LIFETIME, &attributes, (psa_key_lifetime_t)2);
```

With those changes, then the tests are executed using the SE driver and in this example, Parsec.
A few comments however:
* the fact that the test stops as soon as the first failure happens makes it hard to be used if not all key types are supported. It would be nice to continue to the next key data even if the previous one failed so that the PSA Functional Tests could be used to see the percentage of compliance with PSA Crypto API.
* the changes above allow touch the target files except for actual test files. I am not sure on what should be the best approach to modify the Key ID and Lifetime in case a Secure Element driver is used. Maybe there could be a pre-processor macro that conditionally compile the two lines if the code is compiled for the SE target?

kotegowder, 2020-06-24:
The changes seems to be okay except at tests layer.

The tests are exercising those items that are listed in specification. 
The changes proposed for key_id and key_lifetime is something that cannot be added in tests.

Maybe more brainstorming needed on this.

jk-arm, 2020-11-03:
closing as the reason stated by vinay is valid for system compliance.

closed 2020-11-03 by jk-arm