ARM-software/psa-arch-tests #223: Tests fail to build on macOS

Patater, 2020-06-23:
The PSA Arch Tests fail to build on macOS.

```
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15: [ 40%] Building C object CMakeFiles/TargetConfigGen.dir/Users/user/working/psa/mbed-os-tf-m-regression-tests/mbed-os/features/FEATURE_EXPERIMENTAL_API/FEATURE_PSA/TARGET_TFM/TARGET_IGNORE/psa-arch-tests/api-tests/BUILD/targetConfigGen.c.o..
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15: /Users/user/working/psa/mbed-os-tf-m-regression-tests/mbed-os/features/FEATURE_EXPERIMENTAL_API/FEATURE_PSA/TARGET_TFM/TARGET_IGNORE/psa-arch-tests/api-tests/BUILD/targetConfigGen.c:1:10: fatal error: 'stdio.h' file not found..
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15: #include <stdio.h>..
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15:          ^~~~~~~~~..
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15: 1 error generated...
[Test-Target] 11:04:15: [Build-PSA-Compliance-tests] 11:04:15: make[5]: *** [CMakeFiles/TargetConfigGen.dir/Users/user/working/psa/mbed-os-tf-m-regression-tests/mbed-os/features/FEATURE_EXPERIMENTAL_API/FEATURE_PSA/TARGET_TFM/TARGET_IGNORE/psa-arch-tests/api-tests/BUILD/targetConfigGen.c.o] Error 1..
```

Since macOS version 10.15 (Catalina), macOS can no longer run 32-bit binaries. The PSA Arch Tests seem to require the ability to build (and perhaps also run) 32-bit binaries on the host (not target), as discussed in https://github.com/ARM-software/psa-arch-tests/issues/162. 

Host: macOS 10.15 64-bit with default system compiler (clang)
Target: Musca B1 with GCC ARM compiler

kotegowder, 2020-06-24:
@Patater : The task of getting away with 32-bit binary generation is there in out backlog. 
We may need to prioritize it among other task we have at hand currently.

Give us sometime to come back on this.

Patater, 2020-08-06:
Removing lines 66 and 67 from `api-tests/tools/scripts/target_cfg/CMakeLists.txt`, allows me to build on Mac.

```diff
 endforeach()
- set_target_properties(${PROJECT_NAME} PROPERTIES COMPILE_FLAGS "-m32 -w")
- set_target_properties(${PROJECT_NAME} PROPERTIES LINK_FLAGS "-m32")
```

closed 2020-08-13 by kotegowder

LDong-Arm, 2020-12-16:
This issues comes back on macOS