ARM-software/psa-arch-tests #226: Cryto Suite Test 24, 25, 36, 37, 39 are failed, proposed fix

AndreyButokNXP, 2020-07-24:
Cryto Suite Test 24, 25, 36, 37, 39 are failed for:
TFM:  TF-M v1.1-RC2,  2020-07-03 (aacfe93b15e5a7ee51d54ef47e6dd1dfe9c65d27)
PSA Arch Tests:  2020-07-21 (833abe106cd27c1b9a402e7aa1c2891d78833aa1)

**Test 24:**
 
    1) test_c024.c:
        //-- TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(3));
        if(check1[i].expected_status == PSA_ERROR_NOT_SUPPORTED)
        {    
            TEST_ASSERT_EQUAL(status, PSA_ERROR_NOT_SUPPORTED, TEST_CHECKPOINT_NUM(3));
            continue; //NXP
        }
        else
        {
            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(3));
        }
   2)  test_c024.c: Move 9 check before 8

   3) test_dta.h: Replace PSA_ERROR_NOT_PERMITTED by PSA_ERROR_CONNECTION_REFUSED

**Test 25**
Sane actions as for Yest 24.

**Test 36**
test_c036.c:
        //-- TEST_ASSERT_EQUAL(status, PSA_ERROR_BAD_STATE, TEST_CHECKPOINT_NUM(2)); 
        TEST_ASSERT_EQUAL(status, PSA_ERROR_CONNECTION_REFUSED, TEST_CHECKPOINT_NUM(2)); 

**Test 37**
test_c037.c:
        //-- TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(7));
        if(check1[i].expected_status == PSA_ERROR_BAD_STATE)
        {    
            TEST_ASSERT_EQUAL(status, PSA_ERROR_CONNECTION_REFUSED, TEST_CHECKPOINT_NUM(7));
            
           /* Abort a cipher operation */
            status = val->crypto_function(VAL_CRYPTO_CIPHER_ABORT, &operation);
            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(9));

            /* Destroy the key */
            status = val->crypto_function(VAL_CRYPTO_DESTROY_KEY, key_handle);
            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(10));
            continue;
        }
        else
        {
            TEST_ASSERT_EQUAL(status, PSA_SUCCESS, TEST_CHECKPOINT_NUM(3));
        }

**Test 39**
    test_dta.h: Replace PSA_ERROR_INVALID_ARGUMENT by PSA_ERROR_CONNECTION_REFUSED

Guess, you should negotiate with TFM about common behavior and returned codes.

kotegowder, 2020-07-29:
Hi Andrej,

The fix is made available at https://github.com/ARM-software/psa-arch-tests/pull/228.
The changes are tested against AN521 and Musca-B1 targets.

Please give it a try and lets us know.

kotegowder, 2020-08-10:
Closing issue as the fix is made available in master.

closed 2020-08-10 by kotegowder

AndreyButokNXP, 2020-08-10:
I see fixes for 24, 25 and 39. 
But nothing for 36 and 37.

AndreyButokNXP, 2020-08-10:
Hello,
the 24 and 25 are failed with TFM caused of different error codes (mentioned before). 
Are you testing it with TFM? If not, could you add it to the regular process?

kotegowder, 2020-08-10:
For us TF-M is the reference platform and 36/37 are passing for me.

AndreyButokNXP, 2020-08-10:
For me, for TFM I have to use PSA_ERROR_CONNECTION_REFUSED instead some errors.
For example:
    .expected_status            = PSA_ERROR_CONNECTION_REFUSED //NXP PSA_ERROR_NOT_PERMITTED

kotegowder, 2020-08-10:
Details on hashes used:
TF-M: aacfe93b15e5a7ee51d54ef47e6dd1dfe9c65d27
psa-arch-tests : 656685b76e33f16ae2417ca5559771d51e4a63f2
Target : AN521

Result for 36 and 37:

TEST: 236 | DESCRIPTION: Testing crypto symmetric cipher APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_cipher_update - Encrypt - AES CBC_NO_PADDING
[Check 2] Test psa_cipher_update - Encrypt - AES CBC_NO_PADDING (Short in)
[Check 3] Test psa_cipher_update - Encrypt - AES CBC_PKCS7
[Check 4] Test psa_cipher_update - Encrypt - AES CBC_PKCS7 (Short input)
[Check 5] Test psa_cipher_update - Encrypt - AES CTR
[Check 6] Test psa_cipher_update - Encrypt - small output buffer size
[Check 7] Test psa_cipher_update - Decrypt - AES CBC_NO_PADDING
[Check 8] Test psa_cipher_update - Decrypt - AES CBC_NO_PADDING (Short in)
[Check 9] Test psa_cipher_update - Decrypt - AES CBC_PKCS7
[Check 10] Test psa_cipher_update - Decrypt - AES CBC_PKCS7 (Short input)
[Check 11] Test psa_cipher_update - Decrypt - AES CTR
[Check 12] Test psa_cipher_update - Decrypt - small output buffer size
[Check 13] Test psa_cipher_update without cipher setup

TEST RESULT: PASSED

******************************************

TEST: 237 | DESCRIPTION: Testing crypto symmetric cipher APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_cipher_finish - Encrypt - AES CBC_NO_PADDING
[Check 2] Test psa_cipher_finish - Encrypt - AES CBC_NO_PADDING (Short in)
[Check 3] Test psa_cipher_finish - Encrypt - AES CBC_PKCS7
[Check 4] Test psa_cipher_finish - Encrypt - AES CBC_PKCS7 (Short input)
[Check 5] Test psa_cipher_finish - Encrypt - AES CTR
[Check 6] Test psa_cipher_finish - Encrypt - AES CTR (short input)
[Check 7] Test psa_cipher_finish - Encrypt - small output buffer size
[Check 8] Test psa_cipher_finish - Decrypt - AES CBC_NO_PADDING
[Check 9] Test psa_cipher_finish - Decrypt - AES CBC_NO_PADDING (Short in)
[Check 10] Test psa_cipher_update - Decrypt - AES CBC_PKCS7
[Check 11] Test psa_cipher_finish - Decrypt - AES CBC_PKCS7 (Short input)
[Check 12] Test psa_cipher_finish - Decrypt - AES CTR
[Check 13] Test psa_cipher_finish - Decrypt - AES CTR (short input)

TEST RESULT: PASSED

******************************************

************ Crypto Suite Report **********
TOTAL TESTS     : 2
TOTAL PASSED    : 2
TOTAL SIM ERROR : 0
TOTAL FAILED    : 0
TOTAL SKIPPED   : 0
******************************************

AndreyButokNXP, 2020-08-10:
May be we are using different TFMs. 
I use TF-M v1.1, released 2020-07-31 (efc7f29fff0ee406a8e6cc0ae9fe56b8e04c5ae7)

kotegowder, 2020-08-10:
I am on known tag : TF-Mv1.1-RC2

AndreyButokNXP, 2020-08-10:
OK. As I am not going to downgrade, will stay with my fixes.