ARM-software/psa-arch-tests #227: Cryto Suite Test 39 is failed, when RSA is disabled

AndreyButokNXP, 2020-07-24:
When RSA is disabled the Cryto Suite Test 39 is failed:

```
******************************************

TEST: 239 | DESCRIPTION: Testing crypto asymmetric APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_asymmetric_encrypt - Invalid algorithm
        Failed at Checkpoint: 3
        Actual: -134
        Expected: 0

TEST RESULT: FAILED (Error Code=0x1)
```

The proposed fix is to add #ifdef ARCH_TEST_RSA_1024  to the "Test psa_asymmetric_encrypt - Invalid algorithm" test case in test_data.h:

```

#ifdef ARCH_TEST_RSA_1024 //Added
#ifdef ARCH_TEST_SHA256
{
    .test_desc              = "Test psa_asymmetric_encrypt - Invalid algorithm\n",
    .type                   = PSA_KEY_TYPE_RSA_PUBLIC_KEY,
    .data                   = rsa_128_key_data,
    .data_length            = 162,
    .usage_flags            = PSA_KEY_USAGE_ENCRYPT,
    .alg                    = PSA_ALG_SHA_256,
    .input                  = NULL,
    .input_length           = 0,
    .salt                   = NULL,
    .salt_length            = 0,
    .output                 = expected_output,
    .output_size            = BUFFER_SIZE,
    .expected_output_length = 0,
    .expected_status        = PSA_ERROR_CONNECTION_REFUSED //NXP PSA_ERROR_INVALID_ARGUMENT
},
#endif
#endif /* ARCH_TEST_RSA_1024 */

```

kotegowder, 2020-07-29:
Hi Andrej,

The fix is made available at #228.
The changes are tested against AN521 and Musca-B1 targets.

Please give it a try and lets us know.

kotegowder, 2020-08-10:
Closing issue as the fix is made available in master.

closed 2020-08-10 by kotegowder